LaserWeb/LaserWeb4 #1: var laserweb = {}; laserweb.add('raster');

openhardwarecoza, 2016-08-29:
So this issue is for us to keep track of the daunting undertaking of @lautr3k and also requested by @tritao  - to modularise LaserWeb (:

lautr3k, 2016-08-31:
Moved to #2

closed 2016-08-31 by lautr3k

emteeoh, 2016-08-31:
Um. I have one question: does what we want to do require a complete re-write? Or could it be just a refactor? I feel like I missed some conversation somewhere that precipitated LW4's creation.

OKOK, I have two questions, but this second one is, IMHO, academic and not so important:
Effectively, for each re-write, LW has been forked by its own author. I don't think this is wrong per se, but it feels off somehow. Like maybe Peter should have just made a branch of LW3 called "V4" or some such? By forking and starting from scratch, mostly, are the optics wrong? What does it all mean? Why are we here? :)

openhardwarecoza, 2016-08-31:
It started because @lautr3k started adding features and fixing LaserWeb3.
My slack coding style and too many global vars,  and no modules,  got in
the way of his work. .  On the other hand there is @arthurwolf 's Fabrica
project which also needs simple modules.  Many modules from lw can even be
reused there and the other way round (if we make our code modules),  and
lastly one of the contributors is busy reusing some of our code in a pure
nodejs backend and thus also needs it more modular.

emteeoh, 2016-08-31:
Ok, these are all good reasons for moving to a "next gen architecture", but does it need a *rewrite*? Couldn't we just start refactoring code? Localise globals, modularise, and maybe fix your coding style (that seems like a minor issue compared to the others)?

openhardwarecoza, 2016-08-31:
I think that was @lautr3k 's original plan. . I think #2 is throwing some spanners in the plan.

lautr3k, 2016-08-31:
Yes that's right, no need to rewrite all the code. The biggest is done.

emteeoh, 2016-08-31:
Ok, so we're branching by making a new repository then. Coo.

openhardwarecoza, 2016-08-31:
(: thats a laserweb tradition