LaserWeb/LaserWeb4 #3: Build process

emteeoh, 2016-08-31:
Um.. I get an error if I run npm install.  Does it need an npm install?

npm ERR! install Couldn't read dependencies
npm ERR! Linux 3.16.0-4-amd64
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.4.7
npm ERR! npm  v2.15.8
npm ERR! path /home/richard/Documents/LaserWeb4/package.json
npm ERR! code ENOPACKAGEJSON
npm ERR! errno -2
npm ERR! syscall open

npm ERR! package.json ENOENT: no such file or directory, open '/home/richard/Documents/LaserWeb4/package.json'
npm ERR! package.json This is most likely not a problem with npm itself.
npm ERR! package.json npm can't find a package.json file in your current directory.

openhardwarecoza, 2016-08-31:
Lw4 doesnt have a node backend yet.  Just serve the app directory in a
webserver of your choice (or npm install -g http-server ;  http-server app

lautr3k, 2016-08-31:
Let us begin with a simple user interface that can work in a web browser (served by an http server of your choice). Eg the smoothieboard host an http server and can communicate via http. The node.js serveur must be used only for things that can not be done in browser like serial communication.

arthurwolf, 2016-08-31:
Oh yes it'd be really really great if this version of laserweb was able to
run stand-alone served by the smoothieboard directly. I don't think it'd be
too hard to do.

openhardwarecoza, 2016-08-31:
Agreed. Smoothies with ether net,  everyrhing with esp8266s, multiple
serial bridges (node,  or go,  or c) -  we have always done as much as we
can in frontend so the backend is easy to swop for something else.  Now
that we have the theory of the various buffers nailed we can technically
replace the harder to install node backend with something in another
language that compiles to an easy to distribute executable (;

openhardwarecoza, 2016-08-31:
Sorry about all the typos...  New phone :(

emteeoh, 2016-08-31:
Ok. at the moment, I'm just using file://... which works!

closed 2016-08-31 by emteeoh

emteeoh, 2016-09-01:
I presume the comment on issue 2 "I expect cross origin warnings in file://" was intended as a response to my comment here...

In short, it worked without any warnings, but I installed http-server anyhow.