LaserWeb/LaserWeb4 #4: Error 404 on OS terminal OSX

cojarbi, 2016-09-07:
After executing the the app i get the following and repeats every couple of minutes
Error 404: { [Error: ENOENT: no such file or directory, open '/Users/ayahni/LaserWeb4/server/../app/inform']
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: '/Users/ayahni/LaserWeb4/server/../app/inform' }

Ive gone from a clean install and folder, but cant find the location the error refers to. Windows install is fine for me

lautr3k, 2016-09-08:
This is not a bug. Just your browser (plugin, app, or something else) that asks the "inform" page/ressource. As this ressource does not exist, the web server reply with an error 404. 
Try to see in your browser console from where this come from.

I will remove 404 notifications in the console, it is not useful for the user, maybe I'll add it in a log file.

cojarbi, 2016-09-08:
With the latest commit there error doesnt show anymore

closed 2016-09-08 by cojarbi