LaserWeb/LaserWeb4 #6: Alternative to node-serialport

openhardwarecoza, 2016-09-11:
Investigate https://www.npmjs.com/package/browser-serialport

lautr3k, 2016-09-11:
> This library only works in a Chrome Packaged App as this is the only way to get access to the serial ports API in the browser.
Maybe a little too restrictive ?

openhardwarecoza, 2016-09-11:
By itself yes.  But as a third option (node serial port,  http and chrome
app)  -  chrome app would be an easier install for new users (fabkit and
darklylabs particularly)

RoaddogLabs, 2016-09-12:
Google is going to EOL Chrome packaged and hosted apps on platforms other than Chrome in the next 12 -24 months.  http://blog.chromium.org/2016/08/from-chrome-apps-to-web.html

For packaging a node app to install as a desktop/traditional app Electron or nw.js are going to be a better choice long term.

quillford, 2016-12-01:
seems like this can be closed?

openhardwarecoza, 2016-12-01:
Well,  sort of.  The current electron app is now working really well for lw3.  But no comms backend exists for LW4 yet.  Agreed though.  Closing this one. Will open a new ticket to start the comms dev for lw4

closed 2016-12-01 by quillford