LaserWeb/LaserWeb4 #7: Supported hardware

emteeoh, 2016-09-14:
Just a comment, I think we should aim for the same support as LW3, at a minimum. Personally, I'll do my best to support grbl, and I'll occasionally be able to test smoothie.

openhardwarecoza, 2016-09-18:
Seconded.  I also have a TinyG I can send to whoever wants to add that?

openhardwarecoza, 2016-09-18:
Sorry closed by accident

funinthefalls, 2016-09-18:
I would love to help out with TinyG, but I do not think my js coding skills are good enough. I have been studying the commits as much as possible to try and speed up my learni4ng process. For now I think I ned to stick to doing the documentation, let me know when you guys are ready for some LaserWeb4 Wiki pages.

lautr3k, 2016-12-08:
What is the purpose of this issue ? Request feature to support firmware specifics features ? 
Thank you for clarifying.

openhardwarecoza, 2016-12-08:
Adding tinyg and marlin mostly.  Tinyg is already in

openhardwarecoza, 2016-12-11:
Closing this issue: TinyG is now implemented into the backend on LW3 - we'll just port this over once solid

closed 2016-12-11 by openhardwarecoza