LaserWeb/LaserWeb4 #8: self = this?

emteeoh, 2016-09-15:
@lautr3k I'm looking at the com code, and part way through the bind_model method, you assign self=this, and then (inconsistently) use it. Is there any particular reason for using "self", or is it just a stylistic thing?

tbfleming, 2016-09-15:
JavaScript functions, especially callbacks, often clobber ```this```. Example:

```
this.can_connect = ko.computed(function() {
                return !self.connected() && !self.wait_connection();
            });
```

```this``` within that callback doesn't point to the object, but ```self``` does.

emteeoh, 2016-09-15:
Ah! Thanks. That explains a lot.

closed 2016-09-15 by emteeoh

lautr3k, 2016-09-15:
MEACULPA (bis); Another thing that ES6 corrects with Arrow objects.

emteeoh, 2016-09-16:
I am *soooo* glad I asked this question! I ran into a problem that happened because of this, so I created self and it worked! It was like I debugged the problem (with your help!) before writing the code.

tbfleming, 2016-09-16:
You might consider arrow functions for newly-written callbacks like @lautr3k alluded to; ES6 added them to replace the self workaround. Chrome supports it without having to use Babel.

emteeoh, 2016-09-16:
Need to learn about them first! In the meantime, self got me moving forward...

Sent from a something something...