LaserWeb/LaserWeb4 #11: CAM: Tree Based

openhardwarecoza, 2016-09-21:
Opening issue to track progress on cam-rewrite by @tbfleming

tbfleming, 2016-09-21:
Here's probably a good spot to explain my available time: it's unpredictable. A window opened up a couple days ago when a work project hit a waiting period. My window could end later this afternoon, next week, or next month. The next window that opens up is unpredictable; it involves both work and non-work things.

openhardwarecoza, 2016-09-21:
Understood (: and thankful for any gaps.

Always feel free to throw down some tasks to delegate though,  i'll be/find
resources to make it happen: like how in the beginning your few quick tips
on clipperjs helped get the initial CNC cam in place -  that way we can
save you some grunt

openhardwarecoza, 2016-12-11:
Closing this issue: @tbfleming 's CAM is awesome

closed 2016-12-11 by openhardwarecoza