LaserWeb/LaserWeb4 #12: UI Design Challenges

tbfleming, 2016-10-01:
There are things that make it challenging to design a UI which is easy on new users, but flexible for experienced users. We need ideas.

Let's start with a big challenge: there's a many-to-many relationship between CAD objects and CAM operations.

  * For new users and quick operations, it'd be nice to load a drawing, select an operation type, fill in some values, and hit go.
  * For slightly more complex operations, it'd be nice to select some objects on the canvas, select an operation type, fill in some values, and hit go.
  * In both cases, it should be obvious how to edit things.
  * At the extreme end:
     * Multiple operations, each operation has multiple objects
     * Need an obvious way to:
       * Create operations
       * Add objects to operations
       * Reorder operations
       * Enable, disable operations
       * Remove objects from operations. Want to avoid making the user delete operations when objects change.
       * All without creating a UI which is imposing on new users
       * Preferably without any kind of context switch (e.g. an easy tab vs. an advanced tab or the awful Wizard dialogs that Microsoft once promoted)
  * It'd be nice to be able to present the CAD object tree, CAM operation list, operation parameter values, and the explanation diagram all on a single screen.
    * These all relate to each other
    * Modal dialogs are painful on experienced users
    * May need to do some kind of progressive disclosure, but that can backfire badly and become annoying.

openhardwarecoza, 2016-10-01:
https://github.com/openhardwarecoza/LaserWeb3/issues/52 was how far I went with UI on the tree work

cojarbi, 2016-10-02:
Indeed the user experience and workflow that LW has make sense. I did expect the CAM tab to further enhance the current workflow. I do think adding that would cater to a more experienced user. In the end there need to be a good balance and no matter what you design, you will always need to provide users with tutorial, step by step guides, etc. Since there are many that simply don't do click with the US pressented

RoaddogLabs, 2016-10-03:
It boils down to what your user base/use cases are.  In my experience laser engraver users are overwhelmingly non CAM/CNC type users.  They tend to be crafters and makers, particularly at the low end.  Even the pro level engravers/cutters don't use a CAM paradigm until one gets to the large flat bed or solid state lasers.  

For whom is this tool intended?  Is it the users that are traditional engravers and cutters that value simplicity and a gradual learning curve or a comparatively small community of DIY CNC users?   A cursory look at the current use case, relatively new users with low end K40s, shows that while these are DIY machine hackers on a budget, for most the grasp of CNC and CAM is minimal, if any.  Many, if not most of these users have difficulty installing the tools from a command line.

There is an incredible test bed in the Google group.  While directly polling them is not likely to lead to many usable paradigms, observing those use cases, types of user and feedback on where the pain points are for those users in the current rev will go a long way toward guiding a better UI/UX for this rev.

openhardwarecoza, 2016-12-11:
Closing this issue... the cam tree + drag and drop seems fine to me (:

closed 2016-12-11 by openhardwarecoza