LaserWeb/LaserWeb4 #13: porting / contribution lead / guidance

openhardwarecoza, 2016-10-18:
@lautr3k @tbfleming is the backend solid enough yet? I have contributors itching to start working on the new code especially now that is compliant, and its React based, and its not a mess anymore (:

@jorgerobles especially wants to dig in

Even me, I'm sitting here not sure where/how to start

Can you writeup a basic "how to" to get us all going ?

tbfleming, 2016-10-18:
I can't think of a how-to, but here are some notes that may help.

* Application state is in a redux store. 
  * e.g. document tree, operations, settings, current tab
* State is never modified, only replaced. This is critical for undo/redo support. New state shares objects with old state to save memory.
* Reducers are the only thing which can create new state. Actions tell the reducers what to do. Components render the state. They also install event callbacks which create actions and dispatch them to the store. 
* Only objects which can be serialized to/from JSON go in the store, no THREE objects, DOM nodes, image objects, etc. This is critical for state saving and loading.
  * e.g. The document tree represents meshes using an array: [x, y, z, x, y, z, ...] 
  * e.g. The document tree will represent image data in base-64-encoded strings, or some other JSON-compatible form.
* React components convert objects to other forms as needed. Functions in lib/ aid this.
  * e.g. The BufferMesh component converts an array of [x, y, z, x, y, z, ...] into a THREE.mesh with a THREE.BufferGeometry. It regenerates things as needed when the state changes.

tbfleming, 2016-10-18:
Example: loading an SVG file.

* The ```Cam``` component sets up a callback for the file input
* The callback fetches the file, creates a ```loadDocument``` action with the file content, and dispatches it to the store
* The ```documents``` reducer handles the loadDocument action. It looks at the file type, sees that it's ```image/svg+xml```, and passes it to the ```loadSvg``` reducer.
* The ```loadSvg``` reducer uses SnapSvg + functions in lib/ to convert the SVG and add it to a new state.
* The store triggers a UI render.

openhardwarecoza, 2016-10-18:
Awesome,  thanks.  Thats a good enough getting started guideline! And thanks for the example too.  
@jorgerobles does that get you going?

jorgerobles, 2016-10-18:
Yes, I'm currently taking a look to the react/redux thing. It's a bit alien for me, but I've played XCOM a while :P.  I hope get that working soon to get started.

Also, noticed is packaged on node, so anyone who downloads the project should run  npm install/start to get it on fire.

openhardwarecoza, 2016-10-18:
Good point.  The readme needs a section on how to deploy too.  ( @lautr3k webpack basic instructions?)

tbfleming, 2016-10-18:
The package list changes frequently; expect to do an ```npm install``` every time you do a ```git pull``` until things settle down.

tbfleming, 2016-10-18:
One more note: don't commit changes that webpack makes to ```docs/``` and ```dist/```. You have to keep a very careful eye on git; it keeps wanting to commit those files, which will cause problems viewing diffs.

jorgerobles, 2016-10-18:
Noted! Should I add to ignore then?

tbfleming, 2016-10-18:
I wish that would work, but git ignores ignore entries for files which already exist.

jorgerobles, 2016-10-18:
Also, note to collaborators this uses Babel preprocessor. I almost faint when see raw html code inside js the first time.

openhardwarecoza, 2016-10-18:
(: see these are the things that need to be in the readme else we'll scare
off contributors

jorgerobles, 2016-10-19:
Well, having knocking around 12 hours or so, I see a major handicap on development.
Although react/redux is with difference a big improvement over legacy code layout, I realize the development could be affected by the steep learning curve this technologies show.
I'm no newbie in this area although I've missed some frontend knacks last couple of years :)

So, I think we need some links about project tools and technologies to get read before starting, easy enough to not be discouraged. @lautr3k, maybe could recommend some?

- [REDUX/REACT] https://egghead.io/lessons/javascript-redux-the-single-immutable-state-tree

_Currently I'm struggling on how to integrate a 3rd party jquery plugin using npm/webpack (Yes, maybe react is the answer in the end, but you don't want to reinvent the wheel if there's good stuff ready to use) and this is for sure one that most will repeated in time, porting legacy code._

openhardwarecoza, 2016-10-19:
@jorgerobles just out of interest, what are you working on?

jorgerobles, 2016-10-19:
Just dabbling with interface to get familiarized. I've managed to do the sidebar resizable as a Split pane. No more form inputs clipping ;)

lautr3k, 2016-10-19:
@jorgerobles I can not recommend anything, perhaps @tbfleming ...
I have no experience with react/redux/babel/webpack, this is my first attempt. See the discussion here : https://github.com/openhardwarecoza/LaserWeb4/issues/2

lautr3k, 2016-12-08:
A guide for developers is a great idea, but I think a page on the wiki would be more suitable.

closed 2016-12-11 by openhardwarecoza