LaserWeb/LaserWeb4 #15: [Workspace] 3d glitches and misbehaviors

jorgerobles, 2016-10-18:
Hi, I've detected some glitches on 3d rendering, on window resizing (appears a black refresh).
Also, the trackball events are not confined over the canvas, so, clicking some element (document tree, for example) makes camera to change position (Ugh!)

I've tracked issue up to http://stackoverflow.com/questions/13670886/allow-mouse-control-of-three-js-scene-only-when-mouse-is-over-canvas,
Digging into code I've got dead end on components/workspace.js::80

tbfleming, 2016-10-18:
It currently does what the stackoverflow answer says :(

tbfleming, 2016-10-18:
I suspect the answer will be to refactor TrackballControls:
* Extract state and move it to the redux store. It will no longer touch the camera directly.
* Rewrite event handlers to use React's event system.

tbfleming, 2016-10-27:
TrackballControls, and its problems, are now gone because of the switch to regl (#16)

jorgerobles, 2016-12-10:
osx 10.9 and chrome  54.0.2840.98 (64-bit)

[image]

jorgerobles, 2016-12-14:
@tbfleming As far I inspected, camera is undefined on Dom3d mount, so no transform is applied and axis are away from viewport. I suspect a componentWillMount on Workspace should fix that.

tbfleming, 2016-12-14:
It was a bug in Dom3d.componentWillUpdate(nextProps). It used this.props instead of nextProps.

jorgerobles, 2016-12-15:
Gone :) Thanks @tbfleming

closed 2016-12-15 by jorgerobles