LaserWeb/LaserWeb4 #16: Mark Moissette's STL parser and viewer

tbfleming, 2016-10-18:
See https://plus.google.com/u/0/112899217323877236232/posts/VR7yA4Zte1N

We might move from THREE to
* https://github.com/regl-project/regl
* https://github.com/mathisonian/react-regl
* Mark's new parser and viewer

kaosat-dev, 2016-10-18:
Hi, just to chime in !
I cannot recommend Regl enough: it might be new, but after using Three.js for years, but  I found Regl to be a real breath of fresh air: 

- you only use what you need
- you have a far greater level of control (shaders , pipeline etc)
- it is functional oriented, and aims to be as stateless as possible 

For the rest : 
Just as an fyi, the parser I wrote uses streams (Node.js ones) + webworkers, this means 
- fast
- non blocking (up to a point)
- chunk based , so you never blow up your browser with huge memory consumption (except at the end of the parsing when you need to actually display the data)

This requires the initial data to be provided as Readeable stream as well, which has some limtations for binary stl data : XHR does not support binary streams, so you need Fetch (chrome); or various polyfills (works almost out of the box in firefox, a bit less so in Safari)

Sorry for the info dump :) I'll be able to share the  links to the (open source) repos in a few days.

tbfleming, 2016-10-18:
LaserWeb4 is Chrome-only; Fetch is fine.

tbfleming, 2016-10-18:
I look forward to the day when SharedArrayBuffer makes it to Chrome's main release.

tbfleming, 2016-10-18:
Skimming though the docs, Regl's a nice improvement over using webgl directly, like I do in jscut and in my work projects. The downside is that it's not friendly towards developers who don't already know GLSL.

openhardwarecoza, 2016-10-19:
One concern I have with dropping three - is our heavy dependency on using three to break circles into segments to do movements.  For example, we currently open a DXF, circles get drawn as three.circlegeometry.  Later when we generate gcode, we iterate the threejs object.  Each circle consists of a couple "lines" - we iterate through those and write out the gcode...

kaosat-dev, 2016-10-19:
@tbfleming  that is true, it does make you dig deeper into shaders (I don't have insane skills in that area either, but you can learn more easilly with Regl than other WebGL tools I used, at least that was my experience)

@openhardwarecoza good point, perhaps it can be alleviated by some other module in here :
http://stack.gl/packages/ : these are lots of small, specialized modules , which you can cherry pick or adapt, so you might be able to find some alternative in there. I regularly mix & match those to use together with reGL

tbfleming, 2016-10-23:
It looks like I'm going to have to dump react-three-renderer. It ties too much stuff (e.g. z buffer settings) to the scene. THREE doesn't have that limit; it's odd that react-three-renderer imposes it. So, no matter which way we choose on regl vs. THREE; we won't have the nice debugger integration.

tbfleming, 2016-10-25:
dev-regl branch:

[image]

tbfleming, 2016-10-27:
regl is now in master.

openhardwarecoza, 2016-11-27:
Closing this for now, looks like we are totally loving ReGL

closed 2016-11-27 by openhardwarecoza