LaserWeb/LaserWeb4 #22: CNC CAM Operation - Drill options

gndprx, 2016-11-01:
Holes seem to be one of the weak links in many of the freely available CAM packages. Would love to see the ability to peck drill as well as spiral cut holes in an efficient manner.

gndprx, 2016-11-01:
Here is a good visual of a spiral cut hole instead of the traditional pocket cutting method.

[image]

gndprx, 2016-11-01:
It looks like spiral pocket is G12 and G13 commands (CW vs CCW)

tbfleming, 2016-11-02:
Do GRBL and Smoothieware support these?

gndprx, 2016-11-02:
good question. Once I have a smoothie on my CNC I'll have to give it a try. Should be a few more weeks before I have one to swap out my TinyG

openhardwarecoza, 2016-11-02:
@wolfmanjm @chamnit

tbfleming, 2016-11-02:
Does TinyG support them?

arthurwolf, 2016-11-02:
There is a optional module in smoothie you can activate for this
See project homepage

chamnit, 2016-11-02:
G-codes G12 and G13 are not currently support by Grbl, but it's not also supported on LinuxCNC. This is a specific-use canned cycle that isn't all that useful unless you are very proficient at hand-coding your g-code programs. And hand-coding g-code programs really need macros and subroutines to make it worth your time, over using a simple CAM program that generate the same tool path for you with G1 motions automatically.

tbfleming, 2016-11-02:
@chamnit does GRBL support G83?

chamnit, 2016-11-02:
@tbfleming : G83 was on the short-list for the imminent Grbl v1.1 release but it didn't make the cut, along with backlash comp, due to flash and memory constraints. Don't worry, I've got some new stuff coming down the pipeline after v1.1 that will address pro-class machine control. The crux of development is now complete with real-time overrides being installed. It should be smooth sailing from here on out.

lautr3k, 2016-12-08:
> There is a optional module in smoothie you can activate for this
@arthurwolf if you talk about my http://smoothieware.org/drillingcycles module, it support only G80-83 and G98-99 unfortunately not (yet) the spiral cycles G12-13

arthurwolf, 2016-12-08:
@lautr3k The reason we have that module in smoothie is if somebody wants to *manually* drill holes. These Gcodes are there to make "manual" Gcode writing easier. I really think something like CNCweb shouldn't be using those, and just generate the longer form of Gcode for the operation. It take a bit more space in the Gcode file, but it makes it much easier to be compatible with new controllers which do not have this feature ( and with Smoothie if this module is not enabled )

openhardwarecoza, 2016-12-11:
I totally agree with @arthurwolf 
Standard G0/1/2/3 workflows for drilling holes is suitable enough and already in the cam... At least for now

closed 2016-12-11 by openhardwarecoza