LaserWeb/LaserWeb4 #25: Rotary Axis support using true A axis

funinthefalls, 2016-11-05:
@tbfleming , @openhardwarecoza asked me to open this issue here in LW4. From the sounds of things, there is no too too much left to do to implement it, just some trig functions for the A values. I would love to have a working rotary axis solution, there are so many project I want to create that require a working rotary axis. I have already built an inexpensive but well built rotary axis setup, just need the firmware to run it now. I will be pushing the fundraising for your K40, I am sure you would love to be able to use a rotary with yours ;)

tbfleming, 2016-11-05:
I currently have this: http://sherline.com/product/3700-cnc-cnc-ready-4-rotary-table/ paired with http://sherline.com/product/3750-tilting-angle-table/. Will it fit in a K40?

funinthefalls, 2016-11-05:
All rotarys have an issue with the clearance under the nozzle. Even using the lightobjects rotary which was designed for the K40 only allows about a 2" diameter object that will fit under the laser nozzle. Solution is to cut out a big hole in the bottom of the case, raise up the machine on extended sides at the bottom.

gndprx, 2016-11-05:
Will this work for both laser and cnc? I have a rotary on the way that I plan to make portable between the two machines.

tbfleming, 2016-11-05:
Laser raster should be straight forward. I'll have to think about the mill ops and other laser ops.

funinthefalls, 2016-11-10:
@tbfleming  I took a look at the links you posted for the rotary, that thing is a monster for a K40. If you want, I can give you some links for some 3d printed or laser cut rotary attachments that are a bet fit size wise.

tbfleming, 2016-11-10:
@funinthefalls I definitely need recommendations. I created a new thread: https://plus.google.com/101442607030198502072/posts/MFNhLYfoTtC

openhardwarecoza, 2016-12-19:
Good enough to close for now?

tbfleming, 2016-12-19:
So far no one's tested it. May malfunction horribly.

cojarbi, 2016-12-19:
Hope I can get to test this week, having trouble with my board and firmware

openhardwarecoza, 2016-12-19:
(; maiming oneself could be a new issue? Closed as #wontfix

funinthefalls, 2016-12-19:
I am waiting for a stepper motor that gets here tomorrow, will test then.

tbfleming, 2016-12-19:
> (; maiming oneself could be a new issue? Closed as #wontfix

Are you saying we should drop rotary support?

openhardwarecoza, 2016-12-19:
No,  i am saying if something goes wrong (; its a new issue hehe

tbfleming, 2016-12-19:
I don't consider it finished; the approach I used isn't compatible with TinyG. I'm waiting to see how well it works (if at all) with smoothie so I have a baseline to compare against when I move it to a different approach.

openhardwarecoza, 2016-12-19:
Noted.  Keeping issue open :)

gndprx, 2016-12-19:
Definitely keep it open. Rotary is a huge part of my go-forward plans for both CNC and Laser. I have my rotary sitting in the shop waiting for me to rewire everything.

Looking forward to having a nice integrated package :)

tbfleming, 2017-02-10:
No testers; code may have rotted. Might remove it.

FabCreator, 2017-02-10:
Give us a few weeks. Does it hurt to leave it in?

gndprx, 2017-02-11:
I can have my rotary up in a few days assuming I can get a few days in the shop where I don't have other more pressing items. I have everything I need and just need to spend some time wiring.

gndprx, 2017-02-12:
Ok, let me rephrase. I can have my rotary up as soon as I can figure out why I'm not able to drive the stepper.

openhardwarecoza, 2017-02-12:
There is a specific build time switch to enable rotary (or we can bug
@wolfmanjm to give you a built bin)

tbfleming, 2017-06-11:
#304 duplicated this, but I didn't catch the duplication in time. Discussion moved there.

closed 2017-06-11 by tbfleming