LaserWeb/LaserWeb4 #29: UI: Makro Panel

openhardwarecoza, 2016-11-10:
Action: Port / Rewrite functionality from LW3

Present Source: https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/macros.js

Desired result: Popup Makro panel with programmable Gcode macros

openhardwarecoza, 2016-12-11:
I love the work you've done on this! Good idea to have them in the jog panel. I put them into bootstrap panels too so we can collapse as appropriate on small screens

openhardwarecoza, 2016-12-11:
https://github.com/LaserWeb/LaserWeb4/commit/20a7cb50f2e7813f96d54fb22ffb9e2edb2da1cb

jorgerobles, 2016-12-11:
Thanks! There's a component that abstracts BS, react-bootstrap, you could use 

import {Panel} from 'react-bootstrap'

<Panel>
</Panel>

openhardwarecoza, 2016-12-11:
darn, sorry! Will redo!

jorgerobles, 2016-12-11:
It's ok, was a friendly advice, not imperative ;)

closed 2016-12-14 by jorgerobles