LaserWeb/LaserWeb4 #31: Machine Profiles

openhardwarecoza, 2016-11-10:
Long requested functionality:  Saved machine profiles / Material database

Also, this can be used to attract sponsors by including premade profiles, profiles can include customised functionality

jorgerobles, 2016-11-10:
+1000

tbfleming, 2016-11-10:
> +1000

What units is that in?

jorgerobles, 2016-11-10:
ThumbsUps of course ;)

jorgerobles, 2016-11-13:
So go refine a bit more. Proposal:

**Machine Profiles:**
Would the profile be an stored group of settings? If this is it, I imagine go to settings tab, and have a dropdown/modal interface where select one of the stored machine settings with some info about the machine. Then the settings forms will be filled up with that profile (maybe an "r u sure?" nag). I think this way you could still touch the settings freely in case you have a customized-of-the-shelf machine.
On the save settings then have the possibility to tag your settings. Balanced between noobs and expert user.

**Material Database:**
This would affect two sections

  1. **On Settings:** The Edit Material Database needs some space on screen. I don't know if We go for a modal, or go for a whole new section without workspace area. Anyways could show an editable datagrid (excel like... http://adazzle.github.io/react-data-grid/index.html) that displays the Name of the material, an Operation profile tag (Cut, Engrave, etc) and then the Operation Speed, Operation Intensity, Operation Passes, and maybe some more I don't know :smile:

  2. **On CAM:** There would be a wizard icon next to the operation selector, and launch some modal that allows you to select the material/operation tag. Then the operation settings will be overriden, like the way described on Machine Profiles, allowing to tweak for the experts and so on.

<:unicorn:> A coooool feature would be allowing to "cloud update" both the Machine profiles and the Material Database </:unicorn:>

jorgerobles, 2016-11-13:
FYI.. not comitted
[image]

openhardwarecoza, 2016-11-13:
Exactly what i had in mind yes.  Dropdown populates defaults.  Just add to
that a way to save your profile

jorgerobles, 2016-11-13:
And a way to delete too :)
The custom profiles would be stored on LocalStorage and merged with the application hardcoded ones. On future will be sent to the server part for full storage. Deal?

openhardwarecoza, 2016-11-13:
Deal.  As long as it can be backed up from the current backup settings
button (: then localstorage is fine

jorgerobles, 2016-11-13:
I could add a download profiles button. No fuss.

jorgerobles, 2016-11-14:
Go, go, go!

[image]

openhardwarecoza, 2016-11-14:
Please post this under the Development. News on g+!

funinthefalls, 2016-11-14:
Extra helping of awesome sauce on this one!

jorgerobles, 2016-11-15:
Now we need some machine profiles :P. I will close the issue until they come.

closed 2016-11-15 by jorgerobles

openhardwarecoza, 2016-12-01:
@quillford @jorgerobles

openhardwarecoza, 2016-12-01:
@jorgerobles what was the exact plan, where are we hosting the machine profiles? Git repo? 
@quillford wants to code this into LW3 and I advised him to talk to you first.  Since LW4 is still a short way from done I dont really want to put off doing it in LW3, but I also dont want whatever we do to be completely incompatible with LW4 work...

jorgerobles, 2016-12-01:
Good question! (I've no mind of it until now :smile: ) It's a great idea to host on a repo, with default setting url of the "official" repo customizable to other users repos (if they want).
But the idea gets complex if we want them able to commit to the repo. We have to study github rate limits, etc.
The folder option will work better for keep their own profiles and tweaks, but for that we need system access (electron...)

kaosat-dev, 2016-12-01:
Electron way is the easiest I think these are essentially 'plugins' like in the atom editor. So no need to worry about github rates etc

openhardwarecoza, 2016-12-01:
'But the idea gets complex if we want them able to commit to the repo' ...

The idea here isnt the indivisual users pushing their private configs up to the  repo.
Its more about the preconfigured "default" templates, a standard K40+Smoothie, a Emblaser, a Fabcreator profile, etc...)

And after playing with Electron yesterday, I am in love.  Electron it is (no argument there) so we can store local configs in json locally.  But for that initial configuration, it would be nice to obtain a "almost good enough" config from the dropdown

jorgerobles, 2016-12-01:
Ok then, there will be a button/automatic way to get the profiles from repo. In the meantime there's a  data/machine_profiles.json where any contributor can work with. It's a simple dictionary,  prefixing the machine profile id with * prevents to be deleted by error in LW4 interface

openhardwarecoza, 2016-12-01:
Cool stuff

@quillford -  check that out and see if thats suitable for your work too?

(or skip doing that in lw3,  rather tackle more lw4 work so we can complete
port sooner)

quillford, 2016-12-01:
To clarify, the way I understand this should be implemented is: for lw4 we have default/preconfigured profiles the user can load but they also have their own customized ones which are saved locally. for lw3 they can basically use the existing restore functionality and get a sightly easier way to load a preconfigured profile. Is this right?