LaserWeb/LaserWeb4 #33: Load, save entire app state

tbfleming, 2016-11-13:
It'd be nice for users to be able to save everything, including the CAM pane. @jorgerobles since you're already handling load/save of settings and machine profiles, want to do this?

jorgerobles, 2016-11-13:
I'll do my best. If is serializable is saveable. 
There's a component (https://github.com/elgerlambert/redux-localstorage) registered on src/index.js. Please tell me more details, as maybe SVG loaded could not be stored, and some state should not be stored to prevent weird behaviors (like undo history)

tbfleming, 2016-11-13:
Everything in the store is serializable, except maybe undo history. I haven't looked closely at whether undo/redo can be saved.

What should be saved:
* To files: everything except maybe undo/redo
* Local storage: probably not documents, operations, and gcode. These can get too large for local storage (this causes a lot of problems in ChiliPeppr). Also probably not the camera since it may be focused in on a particular document.

jorgerobles, 2016-11-17:
Done!

closed 2016-11-17 by jorgerobles