LaserWeb/LaserWeb4 #36: [Wish] CAM: Vector Shape infill

jorgerobles, 2016-11-16:
Hi, I found that for filled vector shapes could be nice to turn vector to raster, o maybe an infill algorithm choosing density and patterns, like 3d printer infills, if possible.

openhardwarecoza, 2016-11-16:
Well,  Sébastien ( @lautr3k)  posted this morning this teaser on g+
https://plus.google.com/+SébastienMischler-Skarab/posts/cHjE7RZkrog
<https://plus.google.com/+S%C3%A9bastienMischler-Skarab/posts/cHjE7RZkrog>

jorgerobles, 2016-11-16:
Cool, now we will need that laser coloring technology isn't it? 
I've found this by the way http://riccardoscalco.github.io/textures/

openhardwarecoza, 2016-11-16:
Me like!  Has artistic application for sure!

cojarbi, 2016-11-16:
That's supper cool and will surely win some users

⁣

Sent from BlueMail

​

jorgerobles, 2016-11-17:
Hey, meager progress but progress after all.
I've done an experiment of using textures.js with LW4. I'm sure I've done the wrong way, but works somehow. It's on https://github.com/jorgerobles/LaserWeb4/tree/d3patterns

Seems textures.js depends upon d3. I'm not sure, I haven't found any doc about using standalone, maybe is possible. So, I've rendered the svg AND then re-rendered as png with a hack. Then that image will be pass through @lautr3k wonder-app or maybe substitute the path in the workspace to be rasterized. Geniuses out there, I summon you!

[image]

tbfleming, 2016-11-17:
@jorgerobles I looked though your example but didn't see the png hack. Did you use the approach [here](https://bl.ocks.org/biovisualize/8187844) or find a different way?

jorgerobles, 2016-11-17:
Pretty similar, it's located on componentes/playground.js

---
        let xml  = new XMLSerializer().serializeToString(ReactDOM.findDOMNode(this)),
            data = "data:image/svg+xml;base64," + btoa(xml),
            img  = new Image()
            img.setAttribute('src', data)

tbfleming, 2016-11-18:
It looks like it will be easy to import an SVG as an image for raster. Inkscape makes it easy to define custom patterns:

[image]

tbfleming, 2016-11-18:
It looks like I'd have to do some refactoring to support this. The code to best handle this is in the document reducer's loadSvg(). Unfortunately the SVG-to-image hack sets up a callback. Reducers shouldn't set up callbacks.

jorgerobles, 2016-11-18:
I will (try to) take a look to the code to maybe find a better place.  My first idea is to use as an operation so you select the group of vectors to convert to raster, select raster operation as would do if the vectors where an image and then preprocess, using the hack before the gcode raster generation.

jorgerobles, 2016-11-18:
Hmmmm.. https://www.npmjs.com/package/svg-patterns, very similar (inspiredby) textures.js 19kb and no d3 dependencies.. yummy

openhardwarecoza, 2016-11-18:
@lautr3k are you doing anything around this in the optimisation?  Or only on svg parsing?

tbfleming, 2016-11-18:
Ah, OK. I misunderstood. Right now it grabs outlines from SVG files but ignores the fill patterns. I assumed you wanted to keep the SVG fill patterns, but it now looks like you want LW4 to generate new fill patterns instead. Is this what you're asking for?

jorgerobles, 2016-11-18:
Sorry, I did not explain myself at all. There is two different but complementary things:

1. The first is I would like LW4 to be capable of rasterizing a vector. For example, Epilog laser "printer" driver treats thicker than 0.02mm vectors, or filled shapes as a raster. Pick a vector, say "this is a raster operation" and voilá.

2. The byproduct of this idea is to indeed pick a hollow shape (but not limited to) and then generate an infill as slicer software does. That could be achieved by two ways, the vector way and the raster way. I suppose the raster way would be easier if the first idea was implemented.

openhardwarecoza, 2016-11-18:
Well,  theres two old things here. One is a way to raster (solid fill with
horizontal lines)  the insides of vectors (svg paths and dxf).  Patterns is
a cool idea... But,  we already have 'rastering'  so i figured handling
filled svg using that code was easier.  Read through
https://github.com/LaserWeb/LaserWeb3/issues/83 basically done,  just needs
a pretty way to select what you want to do in the UI.

Speaking of the way we currently generate gcode. We iterate over the
threejs object,  extract the lines,  and use that as points (see
threegcode.js in lw3/public/js)  -  it made it easier to have three as a
middle way there,  any loader for three,  and we already had a gcode
generator on the other end.  That made adding stl,  dxf,  svg all easy.
Are we still doing the same? (Three did a magnificent job of breaking arcs
and circles into segments which worked 100% of the time)

The other is a crazy idea from a few months back,  we already use @lautr3k
SLAcer.js to 'slice'  and STL into layers.  Adding some infill,  clipperjs
offset the perimeter,  and you basically have a Javascript based stl slicer
for 3D printing.  @raykholo and @andrewhodel both wanted to play with
that....

openhardwarecoza, 2016-11-18:
Stl slicing https://plus.google.com/+PetervanderWalt/posts/jMvRKXpoGdc

tbfleming, 2016-11-18:
We store paths like this: [[x,y,0,  x,y,0,  x,y,0, ...], ...] (I'll probably remove the 0's). This is the ```positions``` member of the document tree. It has easy conversions to/from json, regl buffers, and Clipper paths. STL will eventually convert to [x,y,z,  x,y,z,  x,y,z,  ...].

Some conversion functions (mesh.js):
* ```elementToPositions```. Converts from SVG. Does curve-to-line conversion as needed.
* ```triangulatePositions```. Converts to triangle mesh. Regl uses this to draw the filled area. Regl could also use this to convert paths to raster.
* ```positionsToClipperPaths```

It's OK to use three to do some conversions, but it's no longer the center point of the app's data.

openhardwarecoza, 2016-11-18:
I'll go dig out the links to old issues,  but heads up from the year past's
lessons:

- converting arcs and circles to segments: threejs did a good job.  Nothing
else did
- svg import: snap.svg failed at about 20% of user files,  the one we use
now,  came from the Lasersaur project,  100% success ratio.  Thats also the
one @lautr3k is busy extending in the Optimisation branch of lw3.  Would
prefer we use that one
- dxf import.  Three-dxf from https://github.com/gdsestimating/three-dxf
has been mostly bulletproof.  Just DXF Spline support missing which
@iceman1979 is working in for us.  It draws arcs/circles as
three.circlegeometry so we will now need to extend these to do the
segmentation in the parser?  Or how do we handle it now?  (of course
wishlist here is proper g2/g3 support)

tbfleming, 2016-11-18:
If you can convert three's segments to Clipper, then converting to ```[[x,y,0, x,y,0, x,y,0, ...], ...]``` won't be a problem.

openhardwarecoza, 2016-11-18:
Cool.  Now lemme stop hijacking this thread (:

tbfleming, 2016-11-18:
Try to find those links. I'd like to see what happens when I run those problematic SVG files through.

openhardwarecoza, 2016-11-18:
https://github.com/LaserWeb/LaserWeb2/issues/29

openhardwarecoza, 2016-11-18:
This one too https://github.com/LaserWeb/LaserWeb2/issues/10 -  trying to
jog my own memory,  might not have been snap.svg itself.  May have been
Chilippeprs code on top of snap.svg too...

tbfleming, 2016-11-18:
[image]

openhardwarecoza, 2016-11-18:
All happy then...  @lautr3k. Just note surrounding your work and chime in?

openhardwarecoza, 2016-11-18:
@tbfleming - @lautr3k is busy doing lots of svg parser stuff here
https://github.com/LaserWeb/LaserWeb3/commits/optimization

tbfleming, 2016-11-18:
Reading through LaserWeb/LaserWeb2#29, it looks like the issues were not with Snap itself. Snap preserves groups and group transforms, so you have to account for that. On the plus side, preserving that made this easy:
[image]

jorgerobles, 2016-11-19:
@openhardwarecoza  you are right, I've seen LW3 #83 and that's one way to do that. The other thing It's closer to this (a known cam software ) but with more infill options.
[image]

openhardwarecoza, 2016-11-19:
And I do like this way more,  since lw3:83 only works on SVG.  Whereas this
would work for svg,  dxf, even sliced stls.

openhardwarecoza, 2016-12-11:
Basic fill is done... by @tbfleming 
I suspect we can close this one? 
Reopen if you feel different patterns is a high priorty

closed 2016-12-11 by openhardwarecoza