LaserWeb/LaserWeb4 #42: [WISH] Vector Layout packer

jorgerobles, 2016-11-30:
A big badass herd of <:unicorn:/> from hell.

This feature should should be composed by:

- auto layout elements (or groups), for fitting the best on the work area (with gutter parameters, optionally)
- detect colinear vectors and compute only one resulting vector to gcode (two identical rectangles put one next the other should render one trace for the overlapping edges.)

openhardwarecoza, 2016-11-30:
See https://github.com/Jack000/SVGnest and @Jack000 which we ask every 6 months in 

Today: https://plus.google.com/+PetervanderWalt/posts/ZH5CHVeyCJC
10 Jun 2016: https://plus.google.com/+PetervanderWalt/posts/8UVUACF3BSt
22 Jan 2016: https://plus.google.com/+ArthurWolf/posts/eVxYxnQCiHp
[image]

Note @tbfleming you might be able to figure out how to use the geometryutils functions in SVGNest

openhardwarecoza, 2016-11-30:
@jack000 - ps now on Version 4 - much better architecture, reGL, react, redux, webpack. Much more professional than when you last looked at it

jorgerobles, 2016-11-30:
I'm currently stunned.. what a great team I've joined!

openhardwarecoza, 2016-11-30:
(: I am amazed at that every day too!

jorgerobles, 2016-12-09:
Hey @tbfleming, @lautr3k, have you seen the parser of SVGNest?https://github.com/Jack000/SVGnest/blob/master/svgparser.js
Is there a way to fit any piece of your code with this? Would be the first step to implement layout packing.

Jack000, 2016-12-10:
<pokes head in>
it looks like you guys already have a dxf/svg parser in laserweb3? All you really need is a polygonal representation of the thing you're trying to nest, a lot of the svgparser.js stuff is just loading svgs and polygonifying beziers.

lautr3k, 2016-12-10:
The SVGNest parser do not support group, use, defs viewBox, etc...

Jack000, 2016-12-10:
yes the svgnest parser strips out groups and other non-closed contours as a first step. This makes sense in the context of nesting but probably not for a CAD system or similar.

openhardwarecoza, 2017-02-28:
For now I'm closing this issue... Looking at the issues in #147 i think implementing SVGnest may just be too hard

closed 2017-02-28 by openhardwarecoza