LaserWeb/LaserWeb4 #43: [Question] Sequential job processing

jorgerobles, 2016-11-30:
It comes from a little problem I had this evening. A malformed gcode spoiled all the work, rendering the cut useless (break in the middle of the work). What would be the best solution?

I think the LW4 workflow does this already, but just to be sure @tbfleming, @openhardwarecoza 
I would wish to select the order the jobs are applied, So create one job for a engraving a shape and other to cut after (an outer shape), and then another job for another shape engrave and cut, and so on.

I could not imagine any other way to do, maybe select all the vectors you want to engrave and use the "create jobs invidually for the (outermost?) selected objects" button. What do you think?

jorgerobles, 2016-11-30:
That spins to vector filter selector .. maybe select all the (#ff0000) vectors and make a job/s with them.

tbfleming, 2016-12-01:
You can create multiple operations in lw4; they execute in order. You control which shapes go with which operations by dragging. If you're using Inkscape, then you can place the objects in different layers in the SVG file, then drag the layers to the appropriate operations.

jorgerobles, 2016-12-01:
Perrrfect

closed 2016-12-01 by jorgerobles