LaserWeb/LaserWeb4 #45: LW4 broken

tbfleming, 2016-12-03:
```
./src/reducers/machine-profiles.js
Module not found: Error: Cannot resolve 'file' or 'directory' ../data/lw.machines/machine-profiles.json
```

jorgerobles, 2016-12-03:
Hi @tbfleming , I've split up the machine-profiles data to a new repository (material-database is on the way too), https://github.com/LaserWeb/lw.machines as @openhardwarecoza suggested (#31)
It should be available as a submodule (https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/.gitmodules)
Sorry I should advertised 😦 
If you cannot get it working (should be easy) I will revert.

tbfleming, 2016-12-03:
Ah-ha. It works fine. If we stick with this, then please update the laserweb-development-environment section of the readme.

A disadvantage to this approach is that it requires two commits for every change, plus extra git commands for every pull. Since we already have to run ```npm install``` frequently, maybe it should go in package.json instead. See https://github.com/LaserWeb/LaserWeb4/blob/005f9f06e86e239d6f62d16675b1d49c9b30875a/package.json#L84

jorgerobles, 2016-12-03:
Ok, no problem, but I would like to keep in the data folder. Is there a way to do that with npm?

tbfleming, 2016-12-03:
I don't think there is. npm forces it under node_modules. An advantage is that it makes it clear to everyone that it lives in a different repository.

openhardwarecoza, 2016-12-03:
In the Electron-LW3 branch we now use some awesome NPM Script fu for that.
It can launch other commands (git init submodules for example).  Say add a
script called setup-dev and have it run npm install; git init submodules;
etc

jorgerobles, 2016-12-03:
Done! I've created a package.json for each repository in case we get a mess with @openhardwarecoza script-fu, and added a "npm run-script installdev" option to obtain my yellow belt.

tbfleming, 2016-12-03:
It needs ```git submodule init``` before ```git submodule update```

jorgerobles, 2016-12-03:
Done!

tbfleming, 2016-12-03:
It works nicely. Now just need to update the readme.

jorgerobles, 2016-12-03:
Sorry :roll_eyes: , and Done

closed 2016-12-03 by tbfleming