LaserWeb/LaserWeb4 #49: Settings Panel number input bugs

tbfleming, 2016-12-04:
[image]

This shows me doing the following:
* Entering ```200``` into Machine Width. Result is 2.
* Entering ```.1``` into Laser Beam Diameter. Result is 1.

I suspect the first problem may because of DOM changes near the input control. The second problem is a React bug; I fixed this problem in the CAM pane today. See NumberInput in components/operation.js for the fix.

tbfleming, 2016-12-04:
@jorgerobles if you want, I can fix the 2nd problem (the React one)

jorgerobles, 2016-12-04:
Thank you todd, please apply your patch. It's probable that also that fixes the first issue.

I have been spending some hours a few days ago around inputs, and I think it would be better to use onBlur, or just "uncontrol" all the inputs as you have done on operation.
Also this enforces my opinion that we should work on core input components to not need to spread code (DRY) on the same bugs.

jorgerobles, 2016-12-04:
As a side issue, We need some data versioning mechanism (data as settings, machines, materials, operations) as  I've encountered localstored data is not valid as we make improvements on the code.
It could be as "easy" as a version flag on each data stores, or as complex as use  JSON Schema, to check the validity of stored data and prune/migrate if needed.

tbfleming, 2016-12-04:
The second problem is now fixed. It didn't solve the first problem.

Please create an issue about data versioning.

commit 808a146c (2016-12-04): Bug fix #49, Material database CSV export (#50)

* CSV WIP
* https://github.com/LaserWeb/LaserWeb4/issues/49
closed 2016-12-04 by jorgerobles