LaserWeb/LaserWeb4 #51: Data versioning mechanism

jorgerobles, 2016-12-04:
Cite from #49:

We need some data versioning mechanism (data as settings, machines, materials, operations) as I've encountered localstored data is not valid as We make improvements on the code.

It could be as easy as a version flag on each data stores, or as complex as use JSON Schema, to check the validity of stored data and prune/migrate if needed.

_this comes as the new pxperinch broke the localstored settings. Now is not important, because We are in early stage development, but could be a real issue when introducing a new feature in the future "golden build"_

tbfleming, 2016-12-04:
Here's something we could try: dispatch a LOAD action after replacing the state with loaded data. Reducers would merge the loaded state into default state to make sure nothing's missing. e.g.

```
export function objectNoId(objectType, initialState) {
        ...
        else if (action.type === 'LOAD')
            return Object.assign({}, initialState, state);
        ...
};
```

jorgerobles, 2016-12-05:
That could work, but you could end with garbage/conflicting state across version.
I was thinking on some kind of hash algorithm. Probably different on different datasets.
On Settings, a checksum (md5) of the dictionary keys. This is easy because Settings state is a one-level dict, and all keys would be present on file/localstore. But for the other... I'm not so sure. Multilevel dictionaries need all the keys to be present to be checked upon with a checksum algo or will fail, unless we have a more robust Schema.

tbfleming, 2016-12-05:
A problem with key hashes is that a large number of them will build up over time. Reducers would have to know what to do for every old hash that ever existed. These will pile up fast when we add new fields and it will be too easy to forget to add them, causing breakage.

Reducers could do this in response to LOAD:
* Set missing fields to defaults
* Delete fields they don't recognize; this happens when we remove old fields.
* If a field changes to an incompatible format (e.g. a {} instead of a []), then it needs a new name. Reducers would detect the old field, delete it, and convert its data to the new field.

jorgerobles, 2016-12-05:
I like that!

jorgerobles, 2016-12-30:
@tbfleming I've found a bug related this patch. Dumped workspace does not reload with Operations id (UUID), so all the operations are the same to the new loaded state (all expanded on click, all their attributes change a the same time, etc)

[escaleras_300-01.json.zip](https://github.com/LaserWeb/LaserWeb4/files/679061/escaleras_300-01.json.zip)

tbfleming, 2016-12-30:
fixed.

jorgerobles, 2016-12-31:
I think it serves well so far.

closed 2016-12-31 by jorgerobles

jorgerobles, 2017-01-01:
Loading workspace in CAM settings throws an error with files not containing some of the attributes.
This occurs because operation component rerenders before "LOADED" event is ever dispatched.
Could a reducer action be triggered without dispatching render? (like a transactional operation in relational databases) @tbfleming 

Scenario:
1. Download attached file [escaleras_300-01.json.zip](https://github.com/LaserWeb/LaserWeb4/files/680093/escaleras_300-01.json.zip)
2. go to CAM pane, choose load workspace.

I suggest to export const {STATEKEY}_DEFAULTSTATE (or initialState) in the reducers, so are "widely" available, then call `loaded({statekey},newState,{STATEKEY}_DEFAULTSTATE)` in the delegated reducer.

tbfleming, 2017-01-01:
Fixed