LaserWeb/LaserWeb4 #52: Integrate rasterizer module

openhardwarecoza, 2016-12-05:
Integrate https://github.com/LaserWeb/lw.Rasterizer.js

tbfleming, 2016-12-05:
@lautr3k do you want to do this, or should I?

lautr3k, 2016-12-08:
I'm going to try.

openhardwarecoza, 2016-12-11:
(: Changed from "feature request" to please code (:  It needs no considering we definately want it in hehe
Whereas feature request label to me means more something like "wishlist"

lautr3k, 2016-12-19:
@tbfleming new module published : `npm install lw.raster-to-gcode`
https://github.com/lautr3k/lw.raster-to-gcode

openhardwarecoza, 2016-12-19:
@lautr3k you rock!

jorgerobles, 2016-12-19:
@lautr3k da beast!

kaosat-dev, 2016-12-19:
wowzers !

tbfleming, 2016-12-19:
@lautr3k loadFromImage() doesn't call _processImage(); I'm going to call it myself temporarily.

tbfleming, 2016-12-19:
Integrated. One issue I know of so far: if you use non-proportional resize, then there's no way to let lw.raster-to-gcode know about it.

[image]

@lautr3k can you replace ```ppi``` with ```xppi``` and ```yppi```?

cojarbi, 2016-12-19:
Is that live?

tbfleming, 2016-12-19:
It is now.

cojarbi, 2016-12-19:
I can't seem to be able to drag the file to operations. It does shows a plus sing

tbfleming, 2016-12-19:
OSX or your Windows VM?

cojarbi, 2016-12-19:
Native Windows 10

tbfleming, 2016-12-19:
Did you try clearing the cache?

tbfleming, 2016-12-19:
Also, let's see if we're on the same version. I'm on Chrome ```55.0.2883.87 m (64-bit)```.

cojarbi, 2016-12-19:
Cache Cleaned
Did not work on Chrome Version 54.0.2840.99
Now working on Version 55.0.2883.87

When i drag there is no visual indicator that i actually have a file picked

tbfleming, 2016-12-19:
I switched D&D from mouse events to pointer events to support touch and pen. 55 added support for these. I still need to add visual indicators.

lautr3k, 2016-12-19:
> loadFromImage() doesn't call _processImage(); I'm going to call it myself temporarily.

I see the problem, I fix it tomorrow.

> can you replace ppi with xppi and yppi?

I'll do it tomorrow too.

lautr3k, 2016-12-20:
@tbfleming package(s) fixed and updated, You no longer need to call `_processImage()` yourself after `loadFromImage()`, and you can now set the ppi settings as `{x, y}` object (or as integer for uniform).

lautr3k, 2016-12-20:
@tbfleming as [requested](https://plus.google.com/u/0/105504973000570609199/posts/S9N9i5TCtK3?sfc=true), here you can find a json file with all settings descriptions https://github.com/lautr3k/lw.raster-to-gcode/blob/master/dist/lw.raster-to-gcode.help.json

tbfleming, 2016-12-20:
updated

jorgerobles, 2016-12-20:
A progress bar could be cool. Actually don't show percentage of progress on console :|

lautr3k, 2016-12-25:
lw.raster-to-gcode@0.5.0 released! Add the Milling feature (multipass)

**Renamed setting:** `beamSize` to `toolDiameter`

**Added settings:**
```
rapidRate: 1500,     // Feed rate in mm/min (G0 F value)

milling  : false, // EXPERIMENTAL
zSafe    : 5,     // Safe Z for fast move
zSurface : 0,     // Usinable surface
zDepth   : -10,   // Z depth (min:white, max:black)
passDepth: 1,     // Pass depth in millimeters
```

[image]

jorgerobles, 2016-12-25:
Santa @lautr3k has come!

tbfleming, 2016-12-25:
Does it compensate for 3D cutter geometry? Ball nose is most common for this type of operation.

lautr3k, 2016-12-25:
Not for now, roughing pass only, I would try to add that for the last pass, but I'm not sure how...

tbfleming, 2016-12-25:
Even roughing passes need cutter comp. Are you familiar with 2D image convolution? There's an image preprocessing step you can apply which is similar, but uses multiply-and-min instead of multiply-and-add to a ball-shaped kernel.

lautr3k, 2016-12-25:
I do not know this technique, but I'm going to find out ;)

lautr3k, 2016-12-25:
@tbfleming If you have any documentation, I will gladly take it.

tbfleming, 2016-12-25:
Sorry I don't.

closed 2017-01-13 by openhardwarecoza