LaserWeb/LaserWeb4 #54: Item bubble size/coordinates not respecting resolution

cojarbi, 2016-12-08:
Using chrome on a 2880x1800 display. Zooming does not fix it either

http://recordit.co/tqpSODf0zG

cojarbi, 2016-12-08:
Maybe its the same issue but as the bubble is not scaling then it gets in the way preventing at some point from seeing whats behind. Imagine the user it trying to align or similar.
[image]

tbfleming, 2016-12-08:
Very odd. Which browser and OS? Here's what it's supposed to do. (Chrome 54 on Windows 10):

[image]

cojarbi, 2016-12-08:
Indeed. As i see the intended above, mine is not sticking to the corner but floating around stuck to the left bottom corner of the path
OSX 10.12
Chrome Version 54.0.2840.98 (64-bit)

tbfleming, 2016-12-08:
Here's what it's currently supposed to do:
* Browser zoom affects all text and controls, except the grid text
* Workspace zoom, pan, and rotate affect grid, paths, images, animated selection border, gcode preview, laser preview
* Upper-right corner of bubble is attached to lower-left corner of the selection, but it doesn't leave the workspace area

cojarbi, 2016-12-08:
Aditional test on OSX
Chromium does not the grid
Safari does not load the page at all

Both have webgl enabled and can open LW3

tbfleming, 2016-12-08:
Chromium surprises me; it should behave exactly like Chrome. Safari is a major PITA to support; Apple's letting much of it rot.

kaosat-dev, 2016-12-08:
Agreed on the sentiment , I can try it & bugfix at least on Safari on mac if that helps @tbfleming

tbfleming, 2016-12-08:
LW4 will target Chrom(ium) only, since it's available everywhere except on devices we won't support anyway (iOS). Any help with the Chrome(ium) issues on OSX would be great!

kaosat-dev, 2016-12-08:
Ok I'll install & take a look

kaosat-dev, 2016-12-08:
Chrome: (54.0.2840.98)[image] 
Chromium:(57.0.2946.0) [image]
So if I get that right, these both have the issue ? Bubble supposed to stay in the corner of the grid right? (same at any scale btw)

kaosat-dev, 2016-12-08:
Just a semi useless fyi , but the problem in Safari is just its lack of support for the 'Fetch' api

kaosat-dev, 2016-12-08:
I'll dig deeper into what causes the issue tomorrow if you don't mind

cojarbi, 2016-12-11:
Not sure if its related but im getting serious computer freezes and lock ups on OSX Chrome. 
[image]

kaosat-dev, 2016-12-11:
@cojarbi do you have a few steps/sample files that you used to reproduce ?

cojarbi, 2016-12-11:
@kaosat-dev bellow the file i was using. Load the file and open the 2 operations as above filling the fields, in less than a minute i get the same result
s
[LaserWeb4.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/644416/LaserWeb4.svg.txt)

kaosat-dev, 2016-12-13:
@cojarbi  ok thanks ! I can at least partially reproduce those issues, memory/CPU is high and crashed chrome when trying to close LW4, will investigate

openhardwarecoza, 2017-01-13:
Is this still a problem?

cojarbi, 2017-01-13:
This should be moved to osx issues 

⁣Ariel Yahni​

cojarbi, 2017-02-28:
I think I have tracked this down to High resolution Retina equiped apple machines.

tbfleming, 2017-06-18:
@cojarbi The fix for #349 might affect this. Could you retest?

cojarbi, 2017-06-18:
Im still getting the same error from yesterday, cant compile or npm start

cojarbi, 2017-06-18:
Folved by copying machine profiles from another instance ( old )

and ITS A MIRACLE, that solved the sizing / clicking on grid

cojarbi, 2017-06-18:
I feel normal again!!!! Tx Sir

closed 2017-07-08 by cojarbi