LaserWeb/LaserWeb4 #57: Pass depth increment on Laser cut operation

cojarbi, 2016-12-08:
For diode Laser mostly a depth increment option should be added.

openhardwarecoza, 2016-12-11:
Indeed a requirement - Emblaser has a Z axis we need to support

jorgerobles, 2016-12-11:
hmm.. We should ping our surgeon @tbfleming for operations :)

tbfleming, 2016-12-11:
It's easy; I just haven't gotten to it yet.

jorgerobles, 2016-12-11:
Z Stage Settings #69 are up an running

[image]

tbfleming, 2016-12-11:
Maybe not so easy? I assumed it'd be like CNC, where user set the work offset so Z=0 is where the focal point hits top of material.

tbfleming, 2016-12-11:
@DarklyLabs How is the emblaser2's Z coordinate going to work? Will it have the work offset so Z=0 is where the focal point hits top of material, or does the CAM need to compute Z based on the options @jorgerobles added?
* Air assist comp (how does this affect Z?)
* Cutting mat comp (is this thickness?)
* Focus height comp (would the machine coordinate system handle this since it's fixed?)
* Default mat thickness (how is this different than cutting mat comp?)

DarklyLabs, 2016-12-11:
@tbfleming 
The Emblaser works very similar to a CNC router, with some extra features simplify use for the end customer.

A Z value of 0 will relate to the laser being focused exactly on the base plate. The focal length of the laser is 50mm, this means there will be a 50mm gap between the bottom of the laser unit and the base plate.

The parameters the customer enters are the following:
**Material_Thickness** (MT): This is how thick their material is they are working with.

**Cutting_Mat_Thickness** (CMT): How thick the cutting mat is. (0 if none is being used)

**Focus_Height_Compensation** (FHC) is a number determined by running a calibration file. It will be a small positive or negative number that helps compensate for any small variations in each machine for perfect focus. This number is usually set only once.

Therefore:
Height to top of material from the base = MT+ CMT + FHC
Maximum z-movement = (MT + CMT + FCH) - 50mm
The Emblaser can handle materials up to 50mm thick.

The Air-Assist is a slightly different case. When the Air-Assist nozzle is installed, the amount of movement downwards from the top of the material is restricted to 10mm. This is to prevent the nozzle colliding with the material. A checkbox for AA_Installed needs to be available.

Therefore:
Maximum z-movement (with AA_Installed) = (MT + CMT + FCH) - 10mm

tbfleming, 2016-12-11:
Does positive Z move the bed down? This is how CNC mills would handle a moving bed (opposite direction of a moving tool head).

DarklyLabs, 2016-12-11:
Our bed is fixed and the laser (tool) head moved up and down.

So positive Z move moves the tool up, further away from the bed.

DarklyLabs, 2016-12-11:
One thing I left off the z-moves in the info above is that they should not go below 0. Since z=0 is the lowest position of the laser head, going below this would cause the stepper to stall and loose its position.

tbfleming, 2016-12-11:
[image]

I tried to generalize this. User provides red values. Z coordinate is focal point.

For Emblazer:
* Max Z = 50
* Cut Z Travel Limit = 50 or 10, depending on if air assist is used. 

No checkbox in the general release since the value it toggles (10) is specific to Emblaser. An Emblaser-specific release could hard code Max Z and set Cut Z based on a checkbox. 

Does this look right?

DarklyLabs, 2016-12-11:
This looks correct.

openhardwarecoza, 2016-12-12:
[image]

Just a maybe, but material thickness probably belongs on the CAM tab, not settings.
Settings should be for things that are rarely changed. Something like material thickness likely changes with every job.  In terms of "defaults" - in LW3 we adopted a "last used autosave" instead (not rolled out to all the fields I wanted to yet) 
I think in LW4 we already do that for G0 feedrate? (Stays at same value between page refreshes) 
So for example If I setup a job at 12mm thick material, next time I open LW4 it defaults back to 12, i change it to 6mm, next time I come back, it defaults to 6mm (that way if you have a couple consecutive jobs on same stock of material you dont need to change it for each)

DarklyLabs, 2016-12-12:
Yes, I agree. 
The material thickness could change every job.
The Cutting Mat thickness, and focus height compensation are changed much less frequently.

openhardwarecoza, 2016-12-12:
@DarklyLabs does the emblaser have homing on Z?  (Does it home to top or bottom?)

@FabCreator - mayb just chime in here if you have requirements too so we discuss the plan suitably for both (:

openhardwarecoza, 2016-12-12:
@darklylabs cutting Mat Fitted can also change between jobs (yes/no
setting)  with its height staying in settings?

DarklyLabs, 2016-12-12:
Homing on the Z is achieved by driving the z axis upwards 60mm. Since there is only 50mm movement this ensures it is always at the top most position.  This position is then set to z=50mm.

DarklyLabs, 2016-12-12:
With respect to cutting mat, I would think this thickness is suitably placed in the settings section. Customers generally keep this consistent.
A value of 0 would indicate no mat installed.

openhardwarecoza, 2016-12-12:
Ahh ok.  Cool then.  I dont know what the mat is for (or not)  thought its
something you install when you cut and remove when you engrave for example
(some commercial machinea have a solid bed for engravings,  and a honeycomb
for cutting -  the solid bed allows a better draft over the work to help
with smoke stains)

DarklyLabs, 2016-12-12:
The mat is the Emblaser equivalent to a honeycomb bed.
It can be left in place for both cutting or engraving. It does take up 10mm of material height so can be removed for customers needing to work on materials that is more than 40mm thick.

jorgerobles, 2016-12-12:
I'm going to strip down the default mat thickness. Should I add a Max Z Travel to allow generic stages? (So maybe another machine has 100mm travel instead 50..)

openhardwarecoza, 2016-12-12:
Whatever we do now would benefit from generic thinking (if the plan is to
attract more machine makers over here in due time)
I've asked @FabCreator to look into this thread as well

jorgerobles, 2016-12-12:
[image]

Not commited yet. I've changed field names to a shorter, generic name. We could add ABBR to popup info balloons.  I will commit if no objections.

FabCreator, 2016-12-12:
Hi all,

The Fabkit works slightly differently we have a moving z bed with approximately 130mm of movement. however knowing these distances are not important for the user for the most part as our machine homes to the material. or in the case of very thin material it will home to the upper Z endstop.
once it homes it bounces off and positions itself 3mm from the head to ensure the perfect focal distance, no need to set material thickness.
However what we would love is a separate home z button (G28Z) 
Another cool feature we would love is to be able to adjust the z depending on what your doing. de-focusing for an engrave can give some interesting effects. As we home to 0 they would need to offset with - values.  Engrave job Z-20, Cutting job Z0 etc (but i think this may have been added looking above)

Im not sure what the air assist max travel does? also what command is the air assist button say? our compressor turns on and off with M106 and M107.

Kind regards,
Bonne

jorgerobles, 2016-12-12:
Air assist is an especial flag used for emblaser Z algorithm (see upper in this thread). We should work further allowing an easy and flexible way to customize vendor requirements.

jorgerobles, 2016-12-12:
Z offset could be added to operation parameters, and in fact automatically reflected on material database operation profiles.

tbfleming, 2016-12-12:
@jorgerobles remove Air Assist; Z Max Travel handles it. Side note: why do all the settings SHOUT LIKE THIS?

I wish Emblaser homed to top of material, that approach is closest to mill setups.

I think this model will work for fab kit with these values:
* Mat Thickness = 0
* Material Thickness = 0 (may confuse users)

We'll also have to drop the Z movement restrictions (Z Max Travel, Cut Z Travel Limit); we don't enforce them on X and Y; why should we enforce them on Z? Emblaser can add Z movement restrictions to a fork.

FabCreator, 2016-12-12:
@tbfleming movement restrictions are not needed. In the fabkit if you go too far a limit switch is hit.
perhaps hard jogging limits can be added as an option in setting but they should be user editable if they are there at all. 

As the fabkit does not require mat thickness or material thickness Is it possible to hide and show different options dependent on profile selected?

The only z feature we need is a focal offset which could be set per operation so you can play with different engraving line thicknesses.

jorgerobles, 2016-12-12:
> @jorgerobles remove Air Assist; Z Max Travel handles it. 

As I've understood, should we drop "Air Assist" and "Air Assist max travel" so Emblaser will reimplement on their fork?  Because "Z Max travel" does not tell the machine not to go down if air assist is connected.

> Side note: why do all the settings SHOUT LIKE THIS?

text-transform:uppercase? I put that on css to increase legibility of smaller font-size. Feel free to remove if anyone is offended by an application setting yelling its name. :grin:

> We'll also have to drop the Z movement restrictions (Z >= 0, Z Max Travel, Cut Z Travel Limit); we don't enforce them on X and Y; why should we enforce them on Z? Emblaser can add Z movement restrictions to a fork.

About Z restrinctions,  Z max travel could be renamed to Machine Depth?

openhardwarecoza, 2016-12-12:
We definitely dont want to maintain two forks.  I suspect what we are
saying here is: mat is just normal offset less 10mm so might as well just
have that profile set it to 40?

FabCreator, 2016-12-12:
@openhardwarecoza is it not possible to address these differences in profiles and settings? check boxes to make different features available etc

tbfleming, 2016-12-12:
> About Z restrinctions, Z max travel could be renamed to Machine Depth?

Renaming doesn't help. Having the Z restrictions at all will break machines which home to the top of the material. Emblaser has an odd setup where Z >= 0. FabKit has a normal milling-style setup where Z <= 0.

openhardwarecoza, 2016-12-12:
I have a very strong belief in the Cartesian standard. If emblaser isnt
following it it might just be an oversight which we can convince them of?

tbfleming, 2016-12-12:
> @openhardwarecoza is it not possible to address these differences in profiles and settings? check boxes to make different features available etc

I'm trying to avoid this since having multiple rule sets complicates the CAM code. I think I have a model which works with both machines, except it won't enforce Emblaser's Z restrictions.

tbfleming, 2016-12-12:
> I have a very strong belief in the Cartesian standard. If emblaser isnt
following it it might just be an oversight which we can convince them of?

Please convince them! Unfortunately it's probably too late; too close to shipping.

openhardwarecoza, 2016-12-12:
Well,  it really only involves a quick change in the smoothie configs (:

At the end of the day I am very much in favour of standards.  If/else cases
is dev hell.  Stick to the standard then your (product here)  works with
every (other product here)

FabCreator, 2016-12-12:
@tbfleming fair enough :) 
I think software limits of axis are best left to the firmware config. i know smoothie has plans for implementing this but i am not sure how far away they are from it working. 
Cant the E2 invert the z axis to make 0 the top and -60 the base? this should correct the issue? currently surly you are jogging your axis inverted anyway??

tbfleming, 2016-12-12:
@openhardwarecoza It means adding a switch to the toolhead which touches off on the material.

tbfleming, 2016-12-12:
@FabCreator The E2's direction is correct. It's their 0 position that complicates things.

tbfleming, 2016-12-12:
Here's an ordering of potential Z=0 positions from best to worst:
* Top of material
* Top of base plate (E2)
* Top of travel (will cause massive confusion)

openhardwarecoza, 2016-12-12:
Ahh ok thought direction was reversed.

As I understand it then, home is to Z max and smoothie is configured with a
max endstop and 50mm axis length.  Homing Z sets workPos to Z50
So if i have a material loaded of 6mm thickness,  and the cutting mat,  my
cut height (if focusing on top of material)  is Z16? As i do a multiple
pass cut i drop down to Z0.5 on the last pass?

openhardwarecoza, 2016-12-12:
(: we should ask Domenic to stay up tomorrow night so we hash this out
without timezone hell

tbfleming, 2016-12-12:
@openhardwarecoza yes.

tbfleming, 2016-12-12:
It's 6am here. I'm online because of sleep troubles.

FabCreator, 2016-12-12:
mmm what about setting Z zero before the start of every job after its been positioned for the material focused and re home at the end of every job?

DarklyLabs, 2016-12-12:
I'm here. All this commotion woke me up!

To catch me up, what is the problem with the z=0 being the lowers position for the laser. Why does it make more sense to set the z=0 at the top of the laser height then go negative?

tbfleming, 2016-12-12:
@FabCreator that'd simplify CAM, but unless E2 adds a touch off sensor that'd be too hard on users.

DarklyLabs, 2016-12-12:
The Emblaser does not have material height detection and hence relies on customers entering their material thickness.
It does not seem unreasonable so I am not sure what this discussion is trying to get us to change?

tbfleming, 2016-12-12:
@DarklyLabs top of laser height would be worse that what you have now. The best user experience would be for a sensor to home Z=0 at top of material. It'd also simplify CAM config. If you can't add a sensor, then keep it as is. We just won't enforce the Z restrictions.

openhardwarecoza, 2016-12-12:
@todd isn't the idea here that homing is irrelevant and start gcode moves
the z offset + mat then sets the WCS to match so work z0=top?

DarklyLabs, 2016-12-12:
@tbfleming Ok. We can add the z-restrictions to our release of the software since is is very specific to our needs.

tbfleming, 2016-12-12:
@openhardwarecoza If the user always used the exact same material thickness.

openhardwarecoza, 2016-12-12:
Thats why i say,  couple lines above,  material height needs to be on cam
tab.  Also what @jorgerobles said,  material height in the material db

DarklyLabs, 2016-12-12:
@openhardwarecoza We can enforce z-homing at the start of every job. It only takes a couple of seconds to perform. Then we can use the WCS idea and the material height etc to set z=0 being the top of the material as you suggest.

tbfleming, 2016-12-12:
@DarklyLabs in that case, material height should move to the jog tab.

DarklyLabs, 2016-12-12:
@tbfleming Do you suggest this to prevent material collision?

tbfleming, 2016-12-12:
@DarklyLabs Traditionally, setting up WCS is a machine control function, not a CAM function. The Jog tab handles machine control.

openhardwarecoza, 2016-12-12:
But in this case we do (already in lw3)  handle wcs for the user.  Laser
users are artists mostly.  They dont need to know what wcs is.  Just that
we do it for them

DarklyLabs, 2016-12-12:
There is a fine line between what is 'traditional' and what makes software easy to use for the end customer. We would hope LW4 was not a 'technical' application to use. There are many of those types of programs out there already.

tbfleming, 2016-12-12:
We could have this in the jog tab:
* Home Z button. Both machines need this.
* A button that sets WCS Z based on various machine settings + a material thickness field. E2 needs this.

[24 more replies]

closed 2017-02-07 by jorgerobles