LaserWeb/LaserWeb4 #58: Inconsistency openning svg

cojarbi, 2016-12-08:
Not sure the reason nor the difference but some open for me some dont.

[CUT LETRAS does not open.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/639446/CUT.LETRAS.does.not.open.svg.txt)
[color boxes does not open.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/639448/color.boxes.does.not.open.svg.txt)
[A does open.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/639450/A.does.open.svg.txt)

lautr3k, 2016-12-08:
Need to port my SVG parser to ES6.

tbfleming, 2016-12-08:
"A": May be another OSX issue:
[image]

"CUTLETRAS" and "color boxes": No path objects. Current parser only handles groups and paths.

@lautr3k does your parser handle non-path objects?

tbfleming, 2016-12-08:
@cojarbi Check Settings -> File Settings -> SVG PX PER INCH. It should have defaulted to 96, but if the defaults didn't initialize properly (e.g. empty or 0) then it may mess up the parser.

lautr3k, 2016-12-08:
@tbfleming my SVG parser curently support all basics shapes (line, polyline, polygone, rect, circle, arc, ellipse), all paths commands, group, viewBox, defs and use.

lautr3k, 2016-12-08:
Just created the repo for the ES6 rewrite as NMP module : https://github.com/lautr3k/lw.svg-parser

openhardwarecoza, 2016-12-08:
(: we also really need the dxf parser to come over.  I still hate svgs (;

cojarbi, 2016-12-08:
@tbfleming Settings are on the default 96

lautr3k, 2016-12-08:
@tbfleming I can not find any more you said the format for the paths...
```
path1 = [
    [x,y,x,y,x,y], // Shape or Line
    [x,y,x,y,x,y]  // Hole if shape
    ...            // Hole...
]
```
```
path2 = [
    [[x,y],[x,y],[x,y]], // Shape or Line
    [[x,y],[x,y],[x,y]]  // Hole if shape
    ...                  // Hole...
]
```
I think path1 ?

tbfleming, 2016-12-08:
path1. It can be a mix of open and closed subpaths and the outer order doesn't matter. e.g. this is valid:
```
path1 = [
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Closed path for outer shape #1
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Closed path for outer shape #2
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    ...
]
```

tbfleming, 2016-12-08:
It uses this for fill and stroke color: ```[r, g, b, a]```

tbfleming, 2016-12-08:
I'm attaching a path stress test. This path object mixes up open and closed paths.

[path stress.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/639792/path.stress.svg.txt)

Here's how it behaves:
[image]

lautr3k, 2016-12-08:
Hum... In my head a path can not contain another path. The only case is to define the shape holes. Otherwise it is a group of paths. I would like to be able to select only one shape at a time.

tbfleming, 2016-12-08:
SVG path objects allow it. It's up to the user to decide when they create them.

tbfleming, 2016-12-08:
If SVG didn't support it, then path union operations would sometimes be invalid.

lautr3k, 2016-12-08:
Yes but if my parser gives you each item separately in a group, is not it better? Anyway that's how it works.

tbfleming, 2016-12-08:
I hate the placement of that stupid close button :)

tbfleming, 2016-12-08:
That would be fine, as long as it doesn't mess up complicated cases. I'll see if I can construct a case that may give grief when breaking up.

tbfleming, 2016-12-08:
[path stress 2.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/639934/path.stress.2.svg.txt)

The existing parser handles this case, but poly2tri hits a limitation. I just pushed a workaround which skips filling the inside when poly2tri barfs. I may have to dump poly2tri for another library.

Here's what it should look like:

[image]

Here's what it actually looks like because of the poly2tri issue:

[image]

* Right way to break it up: turn it into a 7-point path and a 5-point path.
* Wrong way: two triangles.

tbfleming, 2016-12-08:
This test also exposed a bug in CAM that I'll track down.

What it does (wrong):

[image]

What it should do (jscut output):

[image]

lautr3k, 2016-12-08:
Ok, I understand. My parser interprets it as two triangles. I will work on that.

tbfleming, 2016-12-08:
The CAM bugs for Path Stress 2 are now fixed.

tbfleming, 2016-12-08:
Be aware that there's a really nasty numeric stability problem finding intersections within paths. Known algorithms that solve it are complex. Don't try the obvious algebraic equations; they backfire badly with certain border cases. Use Clipper instead.

lautr3k, 2016-12-15:
@tbfleming Just finished, when you have time, look at it
[image]

https://github.com/lautr3k/lw.svg-parser

jorgerobles, 2016-12-15:
Un***kBeleiveable.

openhardwarecoza, 2016-12-15:
Lol!  Yip @lautr3k 's work usually invokes that response (; amazing work
always

tbfleming, 2016-12-15:
@lautr3k es6 promises! nice!

tbfleming, 2016-12-15:
It's now integrated.

cojarbi, 2016-12-15:
+1 Jorge's comment

jorgerobles, 2016-12-15:
Let spoil the happiness, a bit...

`error: TypeError: Cannot read property 'replace' of undefined
    at TagParser._normalizeTagAttr (http://localhost:8080/index.js:132578:40)
    at http://localhost:8080/index.js:132546:32
    at Array.some (native)
    at TagParser._parseTagAttrs (http://localhost:8080/index.js:132540:40)
    at TagParser.parse (http://localhost:8080/index.js:132493:19)
    at Parser._parseElement (http://localhost:8080/index.js:131925:29)
    at http://localhost:8080/index.js:131868:39
    at Parser.parse (http://localhost:8080/index.js:131861:21)
    at http://localhost:8080/index.js:131855:41`

[test_ls3_paths.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/655756/test_ls3_paths.svg.zip)

tbfleming, 2016-12-15:
Did you ```npm run installdev```? If so, you might need to wipe node_modules and rerun it.

tbfleming, 2016-12-15:
Never mind ```installdev```. I just reproduced it.

tbfleming, 2016-12-15:
I hold off pushing to https://laserweb.github.io/LaserWeb4/dist/

tbfleming, 2016-12-15:
test_ls3_paths.svg also fails at https://lautr3k.github.io/lw.svg-parser/dist/example/

jorgerobles, 2016-12-15:
It was running on the previous parser.

lautr3k, 2016-12-15:
Fixed with https://github.com/lautr3k/lw.svg-parser/commit/a6562f5aceb7ab46730dfd49cad6502009f44767 due to trailing ";" on style attribute.

tbfleming, 2016-12-16:
Updated. test_ls3_paths.svg exposes another issue. I use viewBox to transform the coordinates I get from Tag.paths. lw.svg-parser scrambles the viewBox values when these are present:

```
   width="100%"
   height="100%"
```

Please don't modify the viewBox values in lw.svg-parser; if you adjust the viewBox it then it will mess up the transform on my end.

lautr3k, 2016-12-16:
If you want the document Viewox use `parser.document.viewBox` instend of `tag.attr.viewBox` ;) Sorry not yet documented.

lautr3k, 2016-12-16:
After the main SVG was parsed you can access this two properties on the parser instance : 
`
parser.editor         // Editor info { name, version, fingerprint }
parser.document  // Document info { width, height, viewBox } 
                             // where viewBox is { x, y, width, height }
`
https://github.com/LaserWeb/lw.svg-parser/blob/master/src/parser.js#L14-L15

lautr3k, 2016-12-16:
Plus on from the Tag object you get two main functions : 
`tag.getPaths()` return an array of Path objects (all contours + holes)
`tag.getShapes()` return an array of ExPolygons objects from Clipper.js (filled shapes)

lautr3k, 2016-12-16:
Published on NPM ;) `npm install lw.svg-parser`

jorgerobles, 2016-12-16:
Badass!

lautr3k, 2016-12-16:
Updated the `lw.svg-parser` module to v0.2.0 not a big change juste splitted code in smaller modules : 
`lw.svg-curves` and `lw.svg-path`. https://www.npmjs.com/~skarab

tbfleming, 2016-12-16:
What happened to paths? Yesterday the coordinates were in px; today I'm not sure what they are. They don't match px or mm.

lautr3k, 2016-12-17:
I did not touch anything at this level, I just split the code into several modules.

lautr3k, 2016-12-17:
@tbfleming please try to install from NPM `npm install lw.svg-parser --save` simpler for maintenance ;)

tbfleming, 2016-12-17:
@lautr3k It broke when I switched to the NPM package (coordinates messed up), so I didn't commit that.

lautr3k, 2016-12-17:
@tbfleming I think it was already like that before, but you have not seen it. I just saw it on a document with width/height set to 100%. Other documents with a fixed size seems correct... Can you confirm?

tbfleming, 2016-12-17:
Confirmed. It now uses the NPM package.

lautr3k, 2016-12-17:
More generally it crashes on relatives units. I'll try to fix it tomorrow.

lautr3k, 2016-12-17:
Finally done right now ;) Package updated. 

Test file : https://www.w3.org/TR/2011/REC-SVG11-20110816/images/coords/Units.svg
Demo : https://lautr3k.github.io/lw.svg-parser/dist/example/

tbfleming, 2016-12-17:
Confirmed fixed.

tbfleming, 2016-12-19:
@lautr3k Could you preserve image tags? You don't need to do any processing, the ```xlink:href="data:..."``` attribute matches the format I use to store images.

lautr3k, 2016-12-19:
@tbfleming you can access original element via tag.element

tbfleming, 2016-12-19:
@lautr3k it says skips them:

[image]

lautr3k, 2016-12-19:
Haaaa ok, understood, yes of course !

lautr3k, 2016-12-19:
Next year, promised I take English lessons.

lautr3k, 2016-12-20:
Done `<image>` and `<text>` tags ;)

tbfleming, 2016-12-21:
Image integrated:

[image]

cojarbi, 2016-12-21:
Totally killer. Better workflow impossible

maudette01432, 2016-12-21:
This rocks.  Aside from the "laser diameter" not being a tool setting that gets auto populated from machine settings this is a highly workable work flow.  Not much room for improvement on the usability side.

[16 more replies]

closed 2017-01-13 by openhardwarecoza