LaserWeb/LaserWeb4 #60: Setup: Missing dependencies

openhardwarecoza, 2016-12-10:
'npm start' fails with a couple dependencies

react-addons-shallow-compare
unicode

Needed to install both before webpack built the bundle.  Assume missing from package.json

jorgerobles, 2016-12-10:
Doing a grep blame that to "react-select" @tbfleming used to color filter.
I could add to package.json as dependency, but seems that should been resolved by npm a the package install isn't it?

kaosat-dev, 2016-12-10:
yes @jorgerobles : but weirdly enough for me it was for react-toggle ?
Also no complaint about missing unicode package here .

jorgerobles, 2016-12-10:
Ok, doing a full reinstall I've seen a peerDependency I have not seen before. 

> npm WARN react-toggle@2.2.0 requires a peer of react-addons-shallow-compare@~0.14.0 || ^15.0.0 but none was installed.

I haven't seen that before, maybe a change on NPM version o external module change throw that.
About unicode, have been installed without problem

jorgerobles, 2016-12-10:
ouch... there's a hairball here, now react 15.4 is needed :|

kaosat-dev, 2016-12-10:
ouch

kaosat-dev, 2016-12-10:
also , I might ping the author of react-toggle, last I heard peerDependencies are outdated

jorgerobles, 2016-12-10:
Please rm -rf the node_modules and run "npm run installdev". It passed on my machine. Thanks :)

openhardwarecoza, 2016-12-10:
Working now on Win10 (:

closed 2016-12-10 by openhardwarecoza