LaserWeb/LaserWeb4 #62: gitignore missing dist?

openhardwarecoza, 2016-12-10:
Not sure, but I just commited some code (git add -A as usual lazynss) and the /dist/ files got included. 
Shouldnt dist be gitignored?

jorgerobles, 2016-12-10:
uhm, In my opinion, yes it should (#13). First shot on google thrown
http://stephenplusplus.github.io/yeoman.io/deployment.html that tell that moving dist to a subtree is a good idea.

openhardwarecoza, 2016-12-10:
Doh!  @tbfleming says exactly that in
https://github.com/LaserWeb/LaserWeb4/issues/13#issuecomment-254563520
Sorry!  Wont happen again (: 'git add src'  then next time

openhardwarecoza, 2016-12-10:
> Doh!  @tbfleming says exactly that in https://github.com/
> LaserWeb/LaserWeb4/issues/13#issuecomment-254563520
> Sorry!  Wont happen again (: 'git add src'  then next time

http://stackoverflow.com/questions/1274057/how-to-make-git-forget-about-a-file-that-was-tracked-but-is-now-in-gitignore
says there are ways of handling existing files...

jorgerobles, 2016-12-10:
Done and running!

closed 2016-12-10 by jorgerobles

tbfleming, 2016-12-10:
I'll purge the existing generated files from dist, that way .gitignore can do its job well.