LaserWeb/LaserWeb4 #66: Scaling changes origin point

cojarbi, 2016-12-11:
Place a path into some coordinates. Scale it and see the coordinates change as the scaling seem to be to the center. Should it be like this or scale to the origin point?

http://recordit.co/YdiBxdHIer

tbfleming, 2016-12-11:
If it scaled based on origin (x0y0) then it could move a long distance, which could be frustrating, especially to new users.

cojarbi, 2016-12-11:
This way is far easier to center 2 objects between one another but i was trying to get my path into the 0,0 location to reference it to the stock and if i scale then i have to move it again to 0,0

tbfleming, 2016-12-11:
You want it to scale relative to the lower-left point of the selected objects instead of their center?

openhardwarecoza, 2016-12-11:
We had the same workflow in LW3: scale first,  then move (:

cojarbi, 2016-12-11:
Maybe both options?

tbfleming, 2016-12-11:
I'm afraid that adding a bunch of options will confuse new users. Users abandoned Mozilla and switched to Firefox partly because Firefox stripped out most of the options. Eventually the Mozilla org gave up on Mozilla because everyone stopped using it. Firefox became the official replacement.

cojarbi, 2016-12-11:
Agreed. Then maybe need to be part of the documentation

closed 2016-12-11 by cojarbi