LaserWeb/LaserWeb4 #68: DXF Support

openhardwarecoza, 2016-12-11:
Port DXF support from LW3 - however that relied heavily on threejs to split circles/arcs/etc into polylines, so ... probably a hard one
That import lib (three-dxf from https://github.com/gdsestimating/three-dxf) did a fantastic job though... every file we could throw at it worked. Except "lines" dxf wasnt autoclosed. And splines wasnt supported. That was about it

jorgerobles, 2016-12-11:
Could https://github.com/mondalaci/dxf2svg2kicad be of help?

tbfleming, 2016-12-11:
Internally, the new architecture doesn't use SVG; instead it converts SVG to a more convenient format. We don't need DXF to SVG conversion; instead we need DXF to our format.

jorgerobles, 2016-12-11:
So, maybe is a lot of work, but what is the schema of our format? A wiki
page would be cool :D

openhardwarecoza, 2016-12-11:
Have a look in
https://github.com/LaserWeb/LaserWeb3/blob/master/public/lib/dxf/three-dxf.js#L110
Lines are easy enough (from XY to XY extracted)
The bulge geometry function seems easy enough to reuse.  Its the
three.circlegeometry we'll need to abstract to get our own point data

openhardwarecoza, 2016-12-11:
> master/public/lib/dxf/three-dxf.js#L110

@jorgerobles from issue #58:

. It can be a mix of open and closed subpaths and the outer order doesn't
matter. e.g. this is valid:

path1 = [
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Closed path for outer shape #1
    [x,y,x,y,x,y,...], // Closed path for a hole
    [x,y,x,y,x,y,...], // Closed path for outer shape #2
    [x,y,x,y,x,y,...], // Open path
    [x,y,x,y,x,y,...], // Closed path for a hole
    ...
]

tbfleming, 2016-12-11:
The format is also in the readme.

openhardwarecoza, 2016-12-11:
Nb here's @lautr3k 's rewrite of the dxf lib in the lw3 optimisation branch
https://github.com/LaserWeb/LaserWeb3/blob/optimization/public/js/lw/dxf.js

openhardwarecoza, 2016-12-14:
@lautr3k are you still alive (; ?

lautr3k, 2016-12-14:
yess I am alive :) working hard on the SVG parser (ES6 and NPM module) : https://github.com/lautr3k/lw.svg-parser

As I said at the beginning I want to make libraries that everyone can (re)use. Soon the same with the DXF parser.

openhardwarecoza, 2016-12-14:
(; Just checking - got worried about the silence

lautr3k, 2016-12-14:
That said, soon, I start a paid job, I would not be very available for a few weeks.

openhardwarecoza, 2016-12-14:
(: Sad as that is for us, I'm very glad for you!

iceblu3710, 2016-12-30:
I plan on taking a crack at dxf parsing using @lautr3k lw.svg-parser as my roadmap. 

I'm no code ninja but I can probably get support for the Sketchup DXF export options working as I can use the NB70/sketchup-dxf-stl-exporter source only exports 3 line types.

openhardwarecoza, 2016-12-30:
We have a beautiful parser in LW3,  courtesy of
https://github.com/gdsestimating
(so parsing is fine),  just need to swop the threejs circle geometry
modules for @lautr3k 's lw.curves and have the existing parser output the
format as per the readme of lw4 so @tbfleming can catch it to draw.

The existing dxf parser in lw3 already had some extra work (from the
original gdsestimating days)  in that @lautr3k added support for LAYERS to
draw the document  tree in lw3

iceblu3710, 2016-12-30:
Thanks for the clarification. I have a tendency to reinvent the wheel.

I will take a look at the flow this evening and be back with questions.

iceblu3710, 2016-12-31:
Oook. So the end goal is to transform the parsers output into the main programs `state` under the `documents` key.

The object structure is:
```javascript
Object {
  children: Array[], // <= Relational tree of descendants under the same "grouping"
  dataURL: "", // <= N/A
  dpi: 1, // <= N/A
  fillColor: Array[], // <= Obvious
  id: uuid.v4(), // <= Used to track each primitive on canvas
  isRoot: false, // <= true on root of whole object only, not on root of children
  mimeType: "", // <= N/A
  name: "b: bla", // <= name + id or layerName
  rawPaths: Array[], // <= The coords for the primitive
  scale: Array[], // <= FloatingControls modifies this? 
  selected: false, // <= For marching ants outline
  strokeColor: Array[], // <= Obvious
  translate: Array[], // <= FloatingControls modifies this? 
  type: "rect", // <= drawing primitive name
}
```

As it would be nice to do things skeleton and then add complexity I have a simple DXF with a few lines and a box. What are the minimum required fields in the object for the drawing engine to funtion?

```javascript
Object {
  id: uuid.v4(),
  name: "b: bla", // Cam tree layer/group name
  type: "rect", // Cam tree item name
  rawPaths: Array[], // <= The coords for the primitive

  strokeColor: Array[0,0,0,0.8], // Force dark grey 
  scale: Array[1,1,1],
  translate: Array[0,0,0],
}
```

openhardwarecoza, 2016-12-31:
@tbfleming

tbfleming, 2016-12-31:
The entire set of fields is required and will change over time. Create new objects by cloning initialDocument (use ```{...initialDocument}```) then change the necessary fields. There are two cases.

Parent object and group objects: change these fields:
```
let root = {
    ...initialDocument,
    id: uuid.v4(),
    type: 'document',
    name: 'File Name',
    isRoot: true,
    children: [...],
};
```

Shapes:
```
let shape = {
    ...initialDocument,
    id: uuid.v4(),
    type: 'type of object (rect, polyline, etc)',
    name: 'this shows up in UI', // For SVG, I use 'SVG name: SVG id' "name" in SVG is really type (rect, circle, etc.)
    children: [...],
    translate: [0, 0, 0],
    scale: [1, 1, 1],
    rawPaths: [...],
    strokeColor: [r, g, b, a],  // a=1: solid, a=0: invisible
    fillColor: [r, g, b, a],
};
```

Keep a = 0 or 1, except if the source file has alpha, or to indicate a problem. e.g. the SVG importer uses a light transparent grey to indicate both fill and stroke are missing in the original file. This keeps badly-formatted SVG files from being invisible.

iceblu3710, 2016-12-31:
Well I have the dxf parsed and in the reducer ready to map into the new `doc` tree. 

I followed the flow in LW3 and nothing really is the same on the output side, I take it the render and `fillTree()` code is purpose built for each file type loaded? In LW4 we are trying to make a unified docTree that we can just pass to the new draw functions?

I should be able to use the LW3 flow to populate the docTree at a minimum, then I can worry about the point functions.

iceblu3710, 2017-01-01:
I have a working example in PW #113 

Not ready for merge but comments are welcome.

iceblu3710, 2017-01-06:
Well, I have everything (Other than text) in DXF_R12 working. The parser does not support the newer features like splines but the R12 file format saves them as polylines so it's all good.

[TEST_EVERYTHING_R12.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/688867/TEST_EVERYTHING_R12.dxf.txt)
[image]

[image]

Just need to clean up the code, add some documentation and we can put it into beta. I have tried all the dxf's I have so it needs some real world experience now.

closed 2017-01-13 by openhardwarecoza