LaserWeb/LaserWeb4 #70: Import not honoring path size

cojarbi, 2016-12-11:
Not sure if this is a  parser issue

Imported 2 100x100mm squares files , one from illustrator and another from inskscape

Setting SVG per inch changed accordingly

Illustrator 72 per inch will load as 100.35x100.35
Inkscape 90 per inch will load as 99.75x99.75

[100x100squareinkscape.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/644446/100x100squareinkscape.svg.txt)
[100x100 square.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/644447/100x100.square.svg.txt)

tbfleming, 2016-12-11:
Inkscape includes stroke width in measurements, you have to disable stroke to get what you want. Illustrator may do the same.

cojarbi, 2016-12-11:
I can see that now but my files are created only as stroke with no fill ( even if LW4 shows them filled ) Should it just take the whole size into consideration?

openhardwarecoza, 2016-12-11:
(: dont sweat it too much,  the @lautr3k parser handles it correctly.

closed 2016-12-11 by cojarbi

tbfleming, 2016-12-11:
Are you sure it does? It'd have to modify the points in the path.

openhardwarecoza, 2016-12-11:
99.97% certain,  i remember testing a 100mm sq from inkscape and looking at
gcode with X0-X100 moves...  But lets make sure @lautr3k?

cojarbi, 2016-12-11:
Not sure what do you mean but here is a screenshot of the inkscape file 
[image]

tbfleming, 2016-12-11:
@cojarbi Switch to using fill and no stroke.

@openhardwarecoza Here's what Inkscape does. You may be able to compensate rects, but trying to compensate complex paths isn't a good idea.

[image]

openhardwarecoza, 2016-12-11:
@cojarbi just check in lw3 (sorry not at pc anymore)

cojarbi, 2016-12-11:
guys to my surprise its the same in LW3. I did a visual inspection before reporting this and not a gcode one. LW3 shows exactly the same values as LW4.

@tbfleming is LW4 filling the paths by default?

openhardwarecoza, 2016-12-11:
Ok (: my bad! Sorry!

tbfleming, 2016-12-11:
@cojarbi yes

cojarbi, 2016-12-11:
As LW3 show only the final size as was under the wrong impression.

allstar87, 2017-06-30:
> Inkscape includes stroke width in measurements, you have to disable stroke to get what you want. Illustrator may do the same.

How do you do this, and why is Inkscape so badly designed?

tbfleming, 2017-06-30:
> How do you do this, and why is Inkscape so badly designed?

Target audience. Graphic designers often use thick strokes and get annoyed when programs don't include stroke width in measurements. CAD users on the other hand expect tools to not include stroke width in measurements. LW's use of SVG is more like a CAM's use of CAD.

[image]

allstar87, 2017-06-30:
Let me explain what I'm doing in detail: I make keychains in batches of 25.  So I need to draw something once (an ellipse, a circular hole near one end, and a raster image centered in the ellipse) then array copy it.  In inkscape, this is surprisingly labor intensive.  And then, out the other side, import it into LW and it's the wrong size because it doesn't handle line widths consistently?"
So, in order to get this to come out the size I want, I have to use a fill with no stroke.  And a non-white fill so I can see what I'm doing.  And if I set LW to laser cut such a graphic, it will trace the outline?

tbfleming, 2017-06-30:
Yes

allstar87, 2017-06-30:
On second thought, let's not go to Camelot.  Tis a silly place.
Thank you tbfleming, you have been very helpful in reaching understanding of this issue.

tbfleming, 2017-06-30:
Here's a possible approach:
* Solid fill, no stroke on all vectors
* Use different fill colors for different operations
* Create a solid rect with a different fill color to establish spacing (blue in the example below)
* Group it all together
* Use copy/paste and manual positioning to create a rough array

[image]

* Select all
* Object -> Arrange
* Set spacing to 0,0; the solid rects will control the real spacing
* Disable Equal Height and Equal Width
* Set # of rows and columns. Inkscape has a nasty bug here, so you'll have to bounce back and forth between the changing the two fields until it settles down on the values you want.
* Click the Arrange button

[image]

* Create operations in LW4:
  * Include entire document in each operation
  * Laser Raster operation will only get the images
  * Laser Cut operations will only get the vectors. Use the Filter Fill option to grab the right vectors for each op.
  * The color filtering will prevent it from including the spacing rect in the operations.

allstar87, 2017-06-30:
I'm not going to accept a solution that involves pressing CTRL+V 25 times.  What do programmers have against automation?  I have a computer so it can do tedious repetition for me.

allstar87, 2017-06-30:
How do I do this without Inkscape?

jorgerobles, 2017-06-30:
With Illustrator :)
There's no currently support on LW for arrays. You have to do on your
preferred CAD/Vector software.

allstar87, 2017-06-30:
Got a Linux version of Illustrator?
Okay, I just tried this in Inkscape:  I drew a rectangle, 50x30, and put its bottom left corner at 0,0.  I imported a jpg, resized it to 25x20, put it at 0,0 as well.  no copying, grouping, or anything else.  Saved it as an Inkscape SVG.  Imported into LW, the image is still shifted vertically.  Which end is the problem on?

tbfleming, 2017-06-30:
AI doesn't put the px-per-inch value in the SVG header, so you have to change LW's setting to match. I don't remember AI's default value off hand.

openhardwarecoza, 2017-06-30:
@allstar87 see https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/.github/ISSUE_TEMPLATE.md open a proper issue and upload the workspace export etc - if you want help, help us help you

allstar87, 2017-06-30:
Wilco.