LaserWeb/LaserWeb4 #71: gcodeMoveUnits?

tbfleming, 2016-12-11:
@jorgerobles Are there any controllers which use mm/s for F instead of mm/min?

openhardwarecoza, 2016-12-11:
Negative.  Marlin, smoothie,  grbl and tinyg are all mm/min.  The mm/s
laser setting comes from user habits.  Laserdrw,  Corel Plugin and
Moshidraw are all mm/s therefore we eventually switched to mm/s inputs and
x60 before sending the gcode.

jorgerobles, 2016-12-11:
Hmm not as far as I know, but some users like to use mm/s when introducing
the settings. Maybe is not a Gcode setting, but an application one

openhardwarecoza, 2016-12-11:
...  Primarily to ease users who knew what speed they used under the stock
controller

tbfleming, 2016-12-11:
It should be an application setting then.

jorgerobles, 2016-12-11:
Ok :)

2016-12-11 17:06 GMT+01:00 Jorge Robles <jorge.diego.robles@gmail.com>:

jorgerobles, 2016-12-11:
Done!

2016-12-11 17:09 GMT+01:00 Jorge Robles <jorge.diego.robles@gmail.com>:

closed 2016-12-11 by jorgerobles

tbfleming, 2016-12-11:
What does "select" mean? Also a better name would be "Feed Units" (variable: feedUnits)

[image]

jorgerobles, 2016-12-11:
Select is just a placeholder when none was selected (value=""). I could change to any other label too, or strip from the select field, but I experienced some bug on change event if not present.
About the name of the Variable I will change to Feed Units for better understanding.

tbfleming, 2016-12-11:
I used ```react-select``` for Operation; it handles that case nicely.

tbfleming, 2016-12-11:
You'll just have to give it an option to disable the 'X' it adds.