LaserWeb/LaserWeb4 #72: Job order sequence

cojarbi, 2016-12-12:
I assume the current order is based on the order i drag the path. Is/will it possible then to reorder/add a path in the middle ?

tbfleming, 2016-12-12:
Operations execute top to bottom. It reoders the paths within each operation to try to minimize G0 moves. If you need paths to cut in a certain order, then create multiple operations.

tbfleming, 2016-12-12:
Right now there's not an easy way to reorder the operations.

cojarbi, 2016-12-12:
Understood but lets say i forgot or need to add something in the middle as the correct order is not working for me, at this point i would need to delete from that point forward?

tbfleming, 2016-12-12:
For now. Keep this ticket open; it's a feature request.

jorgerobles, 2016-12-12:
@tbfleming will reordering operations in state be enough?

[image]

tbfleming, 2016-12-12:
@jorgerobles yes. Maybe put up/down icons before the X icon.

jorgerobles, 2016-12-12:
Ok, I will take a look

jorgerobles, 2016-12-12:
Done! I wished to use sortables but I could not make them work no matter what.

closed 2016-12-12 by jorgerobles

tbfleming, 2016-12-13:
@jorgerobles 

We need to use something other than ```array.prototype.move```. It modifies a built-in prototype, something that is incredibly dangerous and should only be done by polyfills.

Please stop adding random newlines:
[image]

Please use an autoformatter. I recommend the one that comes built-in to vscode.

[image]

jorgerobles, 2016-12-13:
[Sir, yes, Sir](http://www.firearmsnews.com/files/2011/09/Eastwood.jpg) 😆 
I usually check my commits before for nonsense whitespace. This time was not one of them, though.
Please, tell me: do you have any preferred extensions on VSCode? I use Komodo Edit, and autoformatter has done its work well last 8 years.

tbfleming, 2016-12-13:
I use the Bookmarks extension, the C++ Extension, and changed a bunch of keybindings and settings. The most relevant setting tells the formatter and syntax highlighter to assume .js files are JSX. Maybe Komodo edit's autoformatter will work if it has a setting for JSX.