LaserWeb/LaserWeb4 #73: Identify already added paths to operations on the grid

cojarbi, 2016-12-12:
Once the path is added and multiple path are on the list, there needs to be a ways that the user can click the path and get the selection shown on the grid. Unless the user names each path which already gets read from the file buts highly unlikely the user will.
[image]
 do

jorgerobles, 2016-12-14:
[image]

cojarbi, 2016-12-14:
Exactly. That's solves that and will also help on reordering paths as well

closed 2016-12-15 by jorgerobles