LaserWeb/LaserWeb4 #74: Show indicator of gcode generator in progress

cojarbi, 2016-12-12:
Currently i dont see and indicator running. Generated some Vcarve gcode and thought the browser crashed as it took some time to generate

openhardwarecoza, 2016-12-14:
We used nprogress extensively in LW3 (and instead of for loops that can tak a long time have chrome popup with warnings about unresponsive scripts)  we used 

https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/lib/gcode-viewer/gcode-parser.js#L168-L200

to accurately update the NProgress indicator too (at https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/lib/gcode-viewer/gcode-parser.js#L170)

openhardwarecoza, 2016-12-14:
Similar approach in https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/js/laserraster.js#L338-L356 to update NProgress regularly

tbfleming, 2016-12-14:
I have a hierarchy of functions, some in loops. I'll probably have to move them to a web worker.

jorgerobles, 2017-02-09:
Done on https://github.com/LaserWeb/LaserWeb4/issues/141

closed 2017-02-09 by jorgerobles