LaserWeb/LaserWeb4 #75: List of OSX feature not working

cojarbi, 2016-12-12:
As i test and use LW4 im not sure if the issues are all related or not so i will list here and update accordingly if needed.

V-carve Generation crashes the browser
Clicking path directly on the grid does not select it

jorgerobles, 2016-12-13:
@cojarbi, please attach osx version, web browser, and if possible the svg file you are using, some photo/screencap. 
_Also your tshirt size, and bank account._

cojarbi, 2016-12-13:
OSX 10.12
Chrome Version 54.0.2840.98 (64-bit)

Note that Ive compared all this to Windows 8.1 which works perfectly

Do you need my credit card pin also??

tbfleming, 2016-12-13:
He also wants your passport.

cojarbi, 2016-12-13:
My passport will not be of any good as it analog still

⁣Ariel Yahni​

jorgerobles, 2016-12-13:
Hmm gonna try with Sierra.. On 10.9 works enough (Lags a bit after computation - Vcarve is CPU intensive) Chrome Version 55.0.2883.87 (64-bit)
I have no Path issue neither. 

[image]
[image]

_... My shoes are 9 1/2 ..._

jorgerobles, 2016-12-13:
.. Testing on Sierra and Chrome  55.0.2883.95 (64-bit) works fine.
Sorry @cojarbi, cannot reproduce your issues.

cojarbi, 2016-12-13:
I also see you don't have issue with the bubble. Strange is that I don't get update for Chrome 55.

My pant size is 34

cojarbi, 2016-12-13:
Ok im on Chrome 55 and definitely something is not right. I get multiple lock ups, computer freezes etc. Below some errors from the developer window in case it means anything for you guys

[image]
[image]
[image]
[image]

tbfleming, 2016-12-14:
Does your Mac have a discrete GPU? I wonder if the problem is with integrated graphics.

cojarbi, 2016-12-14:
O yes it does. Please note that i can run under LW4 on VM with Windows 8.1 with no issues

[image]

tbfleming, 2016-12-14:
Dang that's perplexing. There should be more overhead running on a VM. The regl errors are WebGL crashing.

kaosat-dev, 2016-12-18:
General FYI : it seems Chrome messed things up with webgl in Chrome 55, I don't have the same issues as above , but a few Webgl features that even work on safari & mobile (support for LineWidth other than 1 for example )  do not on Chrome55 on Mac anymore

openhardwarecoza, 2017-01-13:
Regl was replaced yes?  @cojarbi is this still an issue?

cojarbi, 2017-01-13:
Haven't tried v carve since then but selecting a path the grid is still not possible

cojarbi, 2017-02-08:
Ok so whom do I need to bribe to get OSX Sierra ( seems ) issues fixed. 

Bubble is driving me crazy, click and drag item on the grid not working.

I do have latest OS, latest video drivers, latest chrome, latest .....

tbfleming, 2017-02-08:
We may have to wait until Google fixes Chrome for Mac and Linux.

openhardwarecoza, 2017-02-08:
Did you remember to sacrifice a kidney?

cojarbi, 2017-02-08:
Damm you Google. Kidney was for OB Makerlinks, I can live without a heart tho

openhardwarecoza, 2017-02-08:
Awww sweet.  #makerlinks.  Bonus to the first person who does a 3 way
corner,  with double miters all 3 beams,  joined with makerlinks...  (;

jorgerobles, 2017-02-08:
@cojarbi Seems Sierra is working well here 😞 

[image]

cojarbi, 2017-02-08:
Nooooooooooooooooo :(. Post your current version please or validate mines

DarklyLabs, 2017-02-08:
We just checked Sierra 10.12.3 and things seem to be working correctly as well.
Perhaps a cache clean may help? It has helped us with isolated problems in the past.

cojarbi, 2017-02-08:
Browser cache is cleaned constantly. But I could try fixing data structure of the entire machine. But it's good to know that it's only me an other clients will not suffer. Do you see resize of the bubble correctly?

DarklyLabs, 2017-02-08:
Yes, it behaves the same as Jorge's gif above.

maudette01432, 2017-02-08:
Except your’s doesn’t render the “T”s as well.  It should say “TUIT” not UI.

-Mike

maudette01432, 2017-02-08:
Where are the two T’s?  It should say TUIT.  Yours mis-renders the same as mine.  Open the SVG in inkscape or look at the ticket.  When I see a RED T and a BLACK UIT then it’ll be good ;)

-Mike

jorgerobles, 2017-02-08:
@maudette01432 the issue of the t does not relate to sierra. Misbehaves as well on mavericks. I've pinged the lw.svg parser author to check the problem with your file. (I have picked your file as I was checking that also) :)

maudette01432, 2017-02-08:
Cool,

Just wanted to make sure.  I’ll open a second ticket for the vCarve operation unless you think it might be somehow linked to the rendering issue (not sure it is since the other operations seem to be good).

-Mike

cojarbi, 2017-02-08:
Mike this is another thread. I think I already logged the vcarve issue

cojarbi, 2017-02-09:
Looking for solution. Are any of you running CUDA drivers?

jorgerobles, 2017-02-09:
Yes, I am

cojarbi, 2017-02-09:
oh come on @jorgerobles give me a brake :). So what else could it be?
Running same OSX, Chrome, video drivers ( what version you are running )

jorgerobles, 2017-02-09:
[image]

@cojarbi I don't know the drivers I'm using, maybe http://www.nvidia.com/download/driverResults.aspx/110844/

cojarbi, 2017-02-09:
ok downgrading the drivers did not work, so its must be something else

jorgerobles, 2017-02-09:
Could you open developer tools on Chrome (console) to see anything abnormal?

cojarbi, 2017-02-09:
http://recordit.co/sJNu9Q39V2

cojarbi, 2017-02-24:
Im going to close this as i havent found anyone else with this issue and seems only related to me. I did start a fresh os with same results

closed 2017-02-24 by cojarbi