LaserWeb/LaserWeb4 #76: Operations: Accordian style?

openhardwarecoza, 2016-12-14:
I little niggle I personaly experience with the operations component:
If I drag a new job down, it get appended to the bottom, so
a) If any operations are already expanded the user will always first need to scroll
b) the user also always has to click the expand button (since there are always unvalued inputs to fill in)

Wouldnt it be better to adopt an accordian style (expanding on operation collapses the others?) and then whn dragging in a new job, default it to expanded (which collapses the others in the accordian, and thus saving the user a scroll and a click?)

tbfleming, 2016-12-17:
done.

openhardwarecoza, 2016-12-19:
[image]

Works awesome thanks!

closed 2016-12-19 by openhardwarecoza