LaserWeb/LaserWeb4 #77: Operations: Drop area is inside scrollable area

openhardwarecoza, 2016-12-14:
The drop target scrolls off screen with a long operation, might be best to place it in another seperator between the document tree and operations so its always visible?

[image]

jorgerobles, 2016-12-14:
Some tweaks to keep that fixed. But I don't know how to use that GetBound black magic to keep the balloons in the correct area, @tbfleming  could you teach us a bit?

tbfleming, 2016-12-14:
I did that to save vertical space. Probably OK to not scroll once the operations are accordianed. To stop it from scrolling, move it up the hierarchy.

jorgerobles, 2016-12-14:
only use className="scaleHalf" (not commited yet) and done :P 

[image]

openhardwarecoza, 2016-12-14:
(: wrong issue.  But i love the look.

jorgerobles, 2016-12-14:
OOPs moooving

jorgerobles, 2016-12-15:
This is done. May I close?

openhardwarecoza, 2016-12-15:
Very happy yes!  Close away!

closed 2016-12-15 by jorgerobles