LaserWeb/LaserWeb4 #78: Swop all checkboxes for toggles

openhardwarecoza, 2016-12-14:
I love the toggles @jorgerobles used in settings
[image]

Any chance we can use them everywhere, for instance the checkboxes in CAM too?
[image]

tbfleming, 2016-12-14:
If they can be shrunk to not waste vertical space

jorgerobles, 2016-12-14:
only use className="scaleHalf" (not commited yet) and done :P 

[image]

tbfleming, 2016-12-14:
@jorgerobles it's not always dispatching the actions.

openhardwarecoza, 2016-12-15:
Initial state of the toggle is wrong.  (Union is performed, though Union toggle is off.  First click turns on Union (CAM still Unions) and Second click finally works correctly (disabling Union in the app)

jorgerobles, 2016-12-15:
Fixed!

openhardwarecoza, 2016-12-15:
Still shows :

[image]

(Implies the job won't be Unioned)
But clicking generate gcode generates gcode which is Unioned

(IE Union = yes was the default state on cam side?  Maybe the logic of the checkbox is reversed? @tbfleming

jorgerobles, 2016-12-15:
[image]
[image]

jorgerobles, 2016-12-15:
[image]

openhardwarecoza, 2016-12-15:
Yeah man! Now its working perfectly!

Thanks!

closed 2016-12-15 by openhardwarecoza