LaserWeb/LaserWeb4 #79: regl issues

tbfleming, 2016-12-15:
Old symptom: WebGL crash after a long period of use.

New symptom: webgl crashes much faster.

Today's Windows 10 update killed my ability to run LW4 on my laptop next to the laser. An update is waiting in the queue for my desktop; I may lose it next. It's easier to debug on my desktop, so I'm going to fetch the update in hope that it will help me track down the underlying cause.

tbfleming, 2016-12-15:
It didn't hurt my desktop. Very odd.

openhardwarecoza, 2016-12-16:
My Win10 desktop PC rebooted itself with auto updates a minute or two ago, and now I'm also in

```
WebGL: CONTEXT_LOST_WEBGL: loseContext: context lost
```

openhardwarecoza, 2016-12-16:
Seems to happen on hot update most often for me

openhardwarecoza, 2016-12-16:
[image]

openhardwarecoza, 2016-12-16:
Not always though, also happens when just entering values into a cam job

[image]

openhardwarecoza, 2016-12-16:
[image]

Mayb this helps, but even ThreeJS ( @mrdoob) (Used in LW3) crashes after the windows update

@mrdoob   in LaserWeb4 we us reGL, but Laserweb3 is threejs.  Looks like Microsoft pushed an update to Win10 breaking webgl on some machines, affecting both reGL and Threejs

tbfleming, 2016-12-16:
Really weird part: my affected machine handles shadertoy.com fine.

openhardwarecoza, 2016-12-16:
I'll go test that on mine after dinner

openhardwarecoza, 2016-12-17:
Couple minutes of Shadertoy and still no crash

openhardwarecoza, 2016-12-17:
@mikolalysenko

mikolalysenko, 2016-12-17:
Do you have a minimal reproducing example?  Maybe it is running out of shader objects?

openhardwarecoza, 2016-12-17:
Sorry @mikolalysenko I'll have to leave that question to @tbfleming 

I did manage to isolate an incident in chrome://gpu log

```
[9080:9156:1217/073815:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A3CF7480]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[9080:9156:1217/073815:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A3CF7480]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[9080:9156:1217/073815:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A3CF7480]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
GpuProcessHostUIShim: The GPU process exited normally. Everything is okay.

//  Note by @openhardwarecoza  ^ = previous instance, and from here down is after a reload and trying the same thing again 

GpuProcessHostUIShim:
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(16349)] : [.DisplayCompositor-00000224A4869250]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[8680:8268:1217/074053:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4869250]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[... cut ...]
[8680:8268:1217/074057:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4868730]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074107:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4868730]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
[8680:8268:1217/074107:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4868730]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture **filtering.**
```

lautr3k, 2016-12-17:
Same for me, Regl crashes, especially when I try to delete files from the list. No visible issues with THREE at this time. Windows (10.0.14393)

[image]

openhardwarecoza, 2016-12-17:
@kaosat-dev

tbfleming, 2016-12-17:
My log message after a crash on my desktop:
```
GpuProcessHostUIShim:
GpuProcessHostUIShim: The GPU process exited normally. Everything is okay.
```

tbfleming, 2016-12-17:
```
[8680:8268:1217/074107:ERROR:gles2_cmd_decoder.cc(9063)] : [.DisplayCompositor-00000224A4868730]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture **filtering.**
```

@openhardwarecoza did you have any images loaded? It only uses textures for images. It does use non-power-of-2, but it doesn't filter them.

openhardwarecoza, 2016-12-17:
Negative.  Just the old homer simpson svg i always test with

tbfleming, 2016-12-17:
I'm starting to think one of the libraries we use might be using WebGL without us knowing and getting it wrong.

tbfleming, 2016-12-17:
My desktop just crashed again. This time I'm seeing the log swamped with texture errors. I didn't have any images loaded.

kaosat-dev, 2016-12-17:
did you try any of the regl examples on the same machine to see if the errors stems from regl or something else ? Like  https://regl-project.github.io/regl/www/gallery/batch.js.html

tbfleming, 2016-12-17:
@mikolalysenko @kaosat-dev the crashes are incredibly sporadic; I haven't been able to create a small reproducable example yet or reproduce the crashes on anything except LW4. 

Here's something I noticed early in development and wrote complicated code to work around; maybe it's related. If you create a regl resource, use it as a uniform (texture) or attribute (buffer), .destroy() it, then use a draw function with different uniforms and attributes, then WebGL complains that the destroyed object is still used.

kaosat-dev, 2016-12-18:
@tbfleming I am looking into it, although I have only mac/linux desktops so I cannot actually reproduce exactly these issues

openhardwarecoza, 2016-12-18:
@kaosat-dev mentioned in other thread:
`General FYI : it seems Chrome messed things up with webgl in Chrome 55, I don't have the same issues as above , but a few Webgl features that even work on safari & mobile (support for LineWidth other than 1 for example ) do not on Chrome55 on Mac anymore`

Well, thats what I'm on:

[image]

jorgerobles, 2016-12-18:
Hmm what version is the last stable, and what is used in electron?
I'm sure this is not the best option but...
As far as I know electron uses its own compilation of chromium, apart from
the chrome installed on the machine. We could always develop through
electron.

kaosat-dev, 2016-12-18:
@jorgerobles Tomorow I'll try to package a minimalist Regl app in an electron app, this way we can at least eliminate some of the source of errors ?

tbfleming, 2016-12-18:
Electron needs to be at least Chromium 55; I used pointer events to support touch and pen. http://caniuse.com/#search=pointer%20events

jorgerobles, 2016-12-19:
The primary objective of doing the electron thing is to detect if ReGL is dependant of Chrome/Chromium version (like @kaosat-dev). Any features (pointer or else) done if this issue is not overcome are futile [like any resistance]  :]

kaosat-dev, 2016-12-19:
Here is the test app https://github.com/kaosat-dev/electron-regl-test
I basically just repackaged the Regl examples gallery as electron app : since it covers pretty much all of the features, if they all work we can investigate other stuff. 
It is based aroud Chrome/Chromium v53 unless I am mistaken.
Can anybody test this on windows to see if there are any issues ?

lautr3k, 2016-12-19:
@kaosat-dev work great!

kaosat-dev, 2016-12-19:
thanks for testing @lautr3k , @tbfleming , @openhardwarecoza , @jorgerobles  how does it look on your end ?

jorgerobles, 2016-12-19:
I've tested all the examples and looks good. No performance issues.

tbfleming, 2016-12-19:
@kaosat-dev Most work. cube.js and sprites.js come up blank. I didn't try all of them; it keeps loosing my place (scroll position) when I go back (ESC).

tbfleming, 2016-12-23:
Related: odd profile results at https://github.com/LaserWeb/LaserWeb4/issues/101#issuecomment-268928987

tbfleming, 2016-12-27:
Dropped regl for native WebGL and reoptimized rendering. Any performance or crashing issues should go into a new ticket.

closed 2016-12-27 by tbfleming