LaserWeb/LaserWeb4 #80: SettingsValidator: Change to Badge over Settings Sidebar?

openhardwarecoza, 2016-12-16:
@jorgerobles 
The SettingsValidator Icon on the CAM tab confused me for a while, till I realised it relates to the Settings tab not the CAM params? (Unless I am still confused? 

If it is, how about making it a notification on the Sidebar menu (If something is wrong, it also gives a user a clue as to where to click)

Like:   http://stackoverflow.com/questions/23790721/overlap-notification-badge-on-glyph-in-bootstrap-3

[image]

But over the icon on the sidebar?

jorgerobles, 2016-12-16:
> Assigned #80 <https://github.com/LaserWeb/LaserWeb4/issues/80> to

No problem. I did that way to get a clue near a disabled gcode button. But
your solution makes perfect sense.

openhardwarecoza, 2016-12-16:
Maybe a tooltip over a disabled gcode button mentioning a Setting is off,
is a worthy thing too. Users can be stupid sometimes

jorgerobles, 2016-12-16:
The balloon seems enough cue :)

openhardwarecoza, 2016-12-17:
I love it! 

I tweaked it a little (made it 20x20px and increased line-height to 22px to center the icon vertically)

[image]

openhardwarecoza, 2016-12-17:
Oh and the tooltip is an extra great bonus since it tells 'em whats wrong

closed 2016-12-17 by openhardwarecoza