LaserWeb/LaserWeb4 #83: Wasted Space

tbfleming, 2016-12-17:
There's a lot of new margins eating up vertical space.

[image]

openhardwarecoza, 2016-12-17:
(; aww man but it looks so good.  Lol

jorgerobles, 2016-12-17:
It looks nice to me. Anyways we could tweak space later with css. I like
MaterializeCSS also, but is waste(rer) if such word exists.

tbfleming, 2016-12-17:
Yes it looks nice, but is looking nice worth forcing users to scroll?

tbfleming, 2016-12-17:
Except the checkmark. That looks confusing instead of nice.

openhardwarecoza, 2016-12-17:
The main reason I did this is to make a clearer differentiation between Documents and Operations:
We had it really nicely polished in LW3 (No scroll needed)
even more so, the previous issue I asked for, to make operations an accordian, that really cuts down scrolling, not the extra 80px or so this wouldve added (;

Checkmark; Again, something we had working nicer in Lw3 (;

History lesson in the evolution of that UI: https://github.com/LaserWeb/LaserWeb3/issues/52

(Less space and checkboxes)
 [image]

tbfleming, 2016-12-17:
LW3 didn't have these in the tab:
* The Tip
* The diagram (it was on a modal)
* The Drag target

These mean there are more vertical margins to deal with.

Checkboxes are horrible for selections. LW3 and jscut were stuck with them because no one coded up a real selection mechanism. LW4 doesn't have that limit.

openhardwarecoza, 2016-12-17:
Alright you win. Will remove all that after dinner.  Sit tight on cam.js,
document.js so we dont have a merge conflict

tbfleming, 2016-12-17:
Just to be clear, the only things I'd like to see removed:
* Some of the vertical margins
* Checkmarks

I like the clarity you got by moving stuff around. I fixed the error bubble clipping to exclude the non-scrolling drag target.

openhardwarecoza, 2016-12-17:
Just about doesnt fit into 1280x720
[image]

But our smallest realistic target is the Pipo X9 on the Fabkit (1920x1200 8.9" display)
Th resolution is quite good but making things fingerable (oops) is harder because of the 8.9" display

As is right now it fits with tonnes of space to spare... 
[image]

But... dragging documents isnt very fingerable on touch (; which brings me back to bargaining for checkboxes

openhardwarecoza, 2016-12-17:
(;  
Playing your own arguments against you (;

[image]

Enabling touch on Windows th "native selection" is checkboxes (;

openhardwarecoza, 2016-12-17:
I'll open seperate issue for Pipo / drag / select

closed 2016-12-17 by openhardwarecoza

tbfleming, 2016-12-17:
Wow that's a change; I've never looked at Windows' touch mode. I thought you once said LW4 wouldn't support touch devices. Something about it opening a can of worms with layout...

openhardwarecoza, 2016-12-17:
I said we wont support a "mobile" interface,ie none of the bootstrap responsive layout crap - thats just hell to make look good (horisontal and vertical layouts, scalable to small screens) 

But the Pipo is a "desktop" with a 8.9" touchscreen ... We had it fully supported in LW3 (play a little with LW3 - there are some history lessons worth learning from.   Settings -> Tool -> Enable Touch Numpads being another)

openhardwarecoza, 2016-12-17:
Opened #86 to continue that discussion....

For now I cleaned up space and removed my FA checkboxes... Will leave you to think of fingerable solutions in #86