LaserWeb/LaserWeb4 #84: fresh clone git submodule issues

openhardwarecoza, 2016-12-17:
```
$ npm install ; npm run installdev
npm ERR! addLocal Could not install C:\Users\peter\Documents\Github\LaserWeb4\lw                                                                                                                                                                                               .svg-parser
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\peter\\AppData\\                                                                                                                                                                                               Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.9.2
npm ERR! npm  v4.0.2
npm ERR! path C:\Users\peter\Documents\Github\LaserWeb4\lw.svg-parser
npm ERR! code ENOENT
npm ERR! errno -4058
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open 'C:\Users\peter\Document                                                                                                                                                                                               s\Github\LaserWeb4\lw.svg-parser'
npm ERR! enoent ENOENT: no such file or directory, open 'C:\Users\peter\Document                                                                                                                                                                                               s\Github\LaserWeb4\lw.svg-parser'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\peter\Documents\Github\LaserWeb4\npm-debug.log

> laserweb@0.4.0 installdev C:\Users\peter\Documents\Github\LaserWeb4
> git submodule init && git submodule update --remote && npm install

Submodule 'src/data/lw.machines' (https://github.com/LaserWeb/lw.machines) regis                                                                                                                                                                                               tered for path 'src/data/lw.machines'
Cloning into 'C:/Users/peter/Documents/Github/LaserWeb4/src/data/lw.machines'...
Submodule path 'src/data/lw.machines': checked out 'f32abd1b9276ad35ff6ec3a75bde                                                                                                                                                                                               a9203d54e48a'
npm ERR! addLocal Could not install C:\Users\peter\Documents\Github\LaserWeb4\lw                                                                                                                                                                                               .svg-parser
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\peter\\AppData\\                                                                                                                                                                                               Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.9.2
npm ERR! npm  v4.0.2
npm ERR! path C:\Users\peter\Documents\Github\LaserWeb4\lw.svg-parser
npm ERR! code ENOENT
npm ERR! errno -4058
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open 'C:\Users\peter\Document                                                                                                                                                                                               s\Github\LaserWeb4\lw.svg-parser'
npm ERR! enoent ENOENT: no such file or directory, open 'C:\Users\peter\Document                                                                                                                                                                                               s\Github\LaserWeb4\lw.svg-parser'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\peter\Documents\Github\LaserWeb4\npm-debug.log

npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\peter\\AppData\\                                                                                                                                                                                               Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "run" "installdev"
npm ERR! node v6.9.2
npm ERR! npm  v4.0.2
npm ERR! code ELIFECYCLE
npm ERR! laserweb@0.4.0 installdev: `git submodule init && git submodule update                                                                                                                                                                                                --remote && npm install `
npm ERR! Exit status 4294963238
npm ERR!
npm ERR! Failed at the laserweb@0.4.0 installdev script 'git submodule init && g                                                                                                                                                                                               it submodule update --remote && npm install '.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the laserweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     git submodule init && git submodule update --remote && npm install
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs laserweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls laserweb
npm ERR! There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to in                                                                                                                                                                                               stall?

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\peter\Documents\Github\LaserWeb4\npm-debug.log

```

Gitconfig not updated after switch to npm?

openhardwarecoza, 2016-12-17:
Already fixed!

closed 2016-12-17 by openhardwarecoza

tbfleming, 2016-12-17:
NPM got confused by the existing directory when I ran ```npm install --save-dev lw.svg-parser``` and produced this:

```
     ...
     "json-stringify-pretty-compact": "^1.0.2",
     "lw.svg-parser": "file:lw.svg-parser",
     "node-uuid": "^1.4.7",
     ...
  ```