LaserWeb/LaserWeb4 #85: Pinch to zoom on touch not working

openhardwarecoza, 2016-12-17:
Zooms entire page's viewport, not viewer as expected
We had touch controls via OrbitControls in threejs on lw3  https://github.com/LaserWeb/LaserWeb3/blob/90ebc88c40da29e803f79e26e32b2863cf550667/public/lib/threejs/OrbitControls.js#L12-L14

tbfleming, 2017-02-18:
New behavior:
* Mouse
  * Left button: pan
  * Middle button swipe up/down: zoom
  * Wheel: zoom
  * Right button: rotate
* Touch pad on Surface Book
  * Drag: pan
  * 2-finger swipe up/down: zoom
  * Drag starting from lower-right pad corner: rotate
* Touch screen (hits Chrome bug)
  * Single finger: pan
  * 2 fingers pinch: zoom
  * 3 fingers swipe: rotate

Chrome bug:
* https://bugs.chromium.org/p/chromium/issues/detail?id=347272
* Summary: Chrome often bypasses javascript touch handling and uses its built-in gestures, even
  though we told it not to (touchAction=none). It sometimes zooms and pans the whole page. Worse, 
  this is a weird phone-type zoom and pan; Chrome's increase/decrease zoom commands and its normal
  scrolling don't undo it.

openhardwarecoza, 2017-02-18:
@gndprx -  documentation (:

yuusuf-sallahuddin, 2017-02-18:
@tbfleming The pan/zoom/rotate all works well on my end, however it seems that now when I select an object from the SVG (e.g. a rectangle) & then try to left-click in the floating dimensions/position box it will not allow me to click in there. Instead it just deselects what I have selected & tries to pan the workspace.

tbfleming, 2017-02-18:
Oops. Fixed.

openhardwarecoza, 2017-02-28:
Tested on my Nexbook touchscreen tablet/netbook and looks to me like its working fine now. Closing issue

closed 2017-02-28 by openhardwarecoza