LaserWeb/LaserWeb4 #86: Cannot multiple select or drag documents on pipo

openhardwarecoza, 2016-12-17:
So we spent a lot of time talking Fabcreator into fitting a Pipo X9 onto their machines: http://www.banggood.com/PIPO-X9-Intel-Z3736F-Quad-Core-8_9-Inch-Dual-Boot-Mini-PC-TV-Box-Tablet-p-997056.html?p=HO1614632392201406RX and https://plus.google.com/+BonneWilce/posts/2UnLfyQ1rp4

Now, in LW3 we gave them numeric keypads, large jog buttons, and in the planned treecam we used checkboxes with an "add to job" button to allow some touch ease of use... Dragging and dropping isnt touch friendly...

in LW4 I cannot get any documents into Operations (; deadend

openhardwarecoza, 2016-12-17:
[image]

openhardwarecoza, 2016-12-17:
[image]
[image]

openhardwarecoza, 2016-12-17:
We had this in LaserWeb2 already!  Yikes long ago
[image]

tbfleming, 2016-12-17:
For the CAM pane, maybe this?
* Make touch click act like CTRL+click
* Listen for correct touch events for touch drag to work

openhardwarecoza, 2016-12-17:
We'll have to give it a whirl

Touch event for pan/zoom on the Viewer too.

openhardwarecoza, 2016-12-17:
(; the pipo is awesome by the way - get one for the significant other for xmas, in two weeks when their bored of it move it to the workshop
Since it runs win10 - and its an intel x86 machine even our Electron EXE deploys fine on it

openhardwarecoza, 2016-12-17:
@jorgerobles just tagging you to make sure you also keep touch in mind (;

openhardwarecoza, 2016-12-18:
(;  I would like an "Add to new Operation" button in addition to drag and drop if I had my way too

tbfleming, 2016-12-18:
Touch D&D is ready to try. I still need to add touch support to the splitters and workspace.

openhardwarecoza, 2016-12-18:
Multi select works excellently (:

Drag works but needs some sort of visual clue that you are dragging...

tbfleming, 2016-12-18:
Don't have a cursor to set, so probably need to play games with CSS.

cojarbi, 2016-12-19:
I can confirm touch / multiple files for selecting and drag and drop working on Surface Pro. I can touch buttons on the workspace and images themselves

openhardwarecoza, 2016-12-20:
All working nicely now, even on Android! Closing!

closed 2016-12-20 by openhardwarecoza