LaserWeb/LaserWeb4 #87: "Lead In" option

Pensee, 2016-12-18:
A "Lead in" option would be very nice, for some tools like Plasma cutter.

 + a great thanks to the dev's team for all the amazing work.

openhardwarecoza, 2016-12-18:
See also previous discussion here:
https://github.com/Smoothieware/Smoothieware/issues/841

I would say we can leverage the existing laseron / laseroff code from settings, to allow users to customise their firing sequence., because even on same firmware, the process is very much machine dependant:

Here's a video of my machine (now sold with the move sadly) https://www.youtube.com/watch?v=0ZMn9tS2_II in action

So the basic start stop of a plasma works like this:

###No Torch Height Control + simple chinese inverter

1.  G0 to start of the lead-in
(Because a plasma start leaves some blowout and a large hole (up to 3x the normal kerf) you want to init the arc next to the cut, in some waste material)

Example:  Start in center of hole, cut to radius, then complete circle
[image]

CAM challenge: Algorithm to determine where to lead-in from (circles are easy = center) but such that its outside any other workpiece (always in scrap) 

1.1 Note; May include a Z move to position it at optimal pierce height to start the arc

2. Start the Arc: Here with Smoothie we tend to use a Switch Module with a custom M-Code http://smoothieware.org/switch   or for Grbl we usually tie off the Spindle pin 
In the case of a cheap chinese inverter, this just flips a relay module, closing the wires that usually went to the handheld torch) 
[image]

2.1  Once the arc is triggered, the HF pre-arc usually fires for less than a second or so, before the main arc starts.  On the cheap machines (and since neither grbl or smoothie can talk to a THC yet to read the ARC OK signal) we just rely on a G4 dwell determined with a little trial and error (longer for thicker material) to allow the arc to pierce the metal

2.2  Once pierced (or sufficient time surpassed) usually machines without a drag torch, needs a Z move here, to bring the nozzle closer to the work (pierce height is around 5mm above surface with a 40A chinese plasma, and around 1mm for cutting) 

3.  Then G1 to complete the Lead In and start the cut

4.  Follow the cut (laser inside / outside existing cam will do from here - plasma kerf == laser beam dia

5.  At end of cut, switch off the Arc (custom M-Code or Spindle off) followed by (if not a drag torch) a Z move back to either zClearance or next Pierce height

6.  Repeat from 1

openhardwarecoza, 2016-12-18:
So in LaserOn (we'll rename the field in settings to Tool On or something @jorgerobles  

we'll typically have

```
G0 Z5
M3
G4 P400
G0 Z1
```

And in ToolOff
```
M5
G0 Z5
```

Since CAM can grab that from Settings, we can manage that with user specified or machine profiles

Th only real dev work is the Lead-In (;

openhardwarecoza, 2016-12-18:
Sheetcam does some extra cool lead in and also corner overshoot moves to improve quality even more 

http://ez-plasma.com/cnc-plasma-torches-accessories/cnc-plasma-software/cnc-plasma-software-sheetcam.html

openhardwarecoza, 2016-12-18:
http://torchmate.com/cad-cam-7-8-9/FAQs-Tool-Paths good read as well

jorgerobles, 2016-12-18:
Should do I change the Laser On/Off label to Tool On/Off or also the variable name (gcodeLaserOn.. Off) ? if the later, just note it @tbfleming

openhardwarecoza, 2016-12-18:
Well, we already have more milling operations than laser (;
So we are pretty wll ahead of overusing the word laser.

jorgerobles, 2016-12-18:
Sooo I went ahead anyways :D also transform gcodeToolOn to a TextArea, for any requirements to come

openhardwarecoza, 2016-12-18:
I also went ahead and added PlasmaInside and PlasmaOutside to the Operations Diagram

[image]

jorgerobles, 2016-12-18:
Then also waterjet ;)

openhardwarecoza, 2016-12-18:
Lol! WaterJet and Plasma pretty much are exactly the same - pierce time etc included

jorgerobles, 2016-12-18:
I guessed that :D  I suppose that diagram will suffice.

tbfleming, 2016-12-18:
From those links, it looks like line lead in is better than curve lead in. Line lead in is also easier to implement.

openhardwarecoza, 2016-12-18:
Agreed,  line is fine.  At the feedrate of consumer plasmas especially (the
industrial 300a units run at insane ipm's and the curve makes for less
vibration,  but with a 40A hobbyist plasma you are running around 800mm/min
in 3mm mild steel

Pensee, 2017-06-26:
keeping a eye on it ...

ril3y, 2020-01-18:
Did this ever get implemented?  In could not find plasma operations in current code base.

cprezzi, 2020-01-20:
Depents on if there is someone willing to implement it. The former cam dev has left the project.

Pensee, 2020-01-20:
If only I had the ability to help it, But I am still very interest by this option ...