LaserWeb/LaserWeb4 #88: Minor UI issue?

funinthefalls, 2016-12-18:
In the CAM tab, the red expand arrow points right to a Close X button, I have found myself accidently clicking on it when not paying attention, the arrow draws me right to the X visually.

openhardwarecoza, 2016-12-19:
Placing them anywhere else will be hard... Shall we write it off as a user error (; @jorgerobles

jorgerobles, 2016-12-19:
I could add a confirm nag to ensure no one closes without acknowledge oooor place the delete inside the fold, so you must unfold and then trash.

tbfleming, 2016-12-19:
@jorgerobles please wait until much later before doing that. I frequently generate a bunch of ops then delete a bunch of ops while testing.

tbfleming, 2016-12-19:
Actually, we'll probably won't need that once undo/redo is fully working (all the filters hooked up and undo/redo buttons added to UI to make it obvious)

jorgerobles, 2016-12-19:
You are right. I use the rule of 3 in my work. Until a request arises 3
times a month we won't do anything ;)

funinthefalls, 2016-12-19:
I have no problems closing this issue. Like I said it is more of an accidental close that happens because the chair<-->keyboard interface unit does not always function correctly ;)

closed 2016-12-19 by funinthefalls