LaserWeb/LaserWeb4 #89: Macro Function (I assume WIP?) fireMacroByKeyboard is not defined

openhardwarecoza, 2016-12-18:
Uncaught ReferenceError: fireMacroByKeyboard is not defined
    at Object.handleMacro (laserweb.js:106)
    at LaserWeb.macro (laserweb.js:66)
    at HTMLDocument._onKeyDown (event_handlers.js:70)

jorgerobles, 2016-12-18:
Fixed. It was a refactor not finished. :slightly_frowning_face: 
The macros functions are indeed WIP until we settle down with the comms part. Right now only fires the action to the reducer.

openhardwarecoza, 2016-12-18:
All cool! Thanks

@cprezzi is our Comms Genius (yeah I picked the title)

closed 2016-12-18 by openhardwarecoza

cprezzi, 2016-12-19:
I am still figuring out where/how I have to work with react / redux... (I realy loved the simplicity of LW3)

openhardwarecoza, 2016-12-19:
@cprezzi - its hard, I agree, but I sort of have my head wrapped around React components already and I must say I love the concept (Define a "component" and reuse it everywhere.  React takes a lot of the hard work out of managing plugging in an extra button etc by rendering the component when it get loaded... and templating.  I see why we need it!)

But agreed I also need to talk through the reducers and actions

The redux store is nice compared to the global vars we had in LW1-3

For example: Check this:

[image]

If i can offer some advice, finish up a perfect backend first and we'll make the frontend plug into it
With that I mean, get server.js to wrap comms into websocket and handle events for connect, send gcode, skip queue gcode, overrides and return status info. Ie cleanup and extend exactly what we were talking about?

On Mon, Dec 19, 2016 at 5:56 PM, Claudio Prezzi <notifications@github.com> wrote:
I am still figuring out where/how I have to work with react / redux... (I realy loved the simplicity of LW3)

openhardwarecoza, 2016-12-19:
https://github.com/LaserWeb/LaserWeb4/issues/13 is worth a read too!

cprezzi, 2016-12-19:
Ok. I do the server part first. :)
I will try to define a uniform interface for the client and handle all the firmware / machine interface specific tuff in the server.

cprezzi, 2016-12-19:
Should I integrate that directly in the LW4 repo, or should I add a new one?

openhardwarecoza, 2016-12-19:
(: thats the big dream!

See if you supply a server, documented with which ws events it emits (and sample strings/objects) as well as which events it expects (and sample data) thats going to make the rest sooo easy

In LW3 we already have it quite simple

| Events Listening  | Events Emitted |
| ------------- | ------------- |
|  socket.on('firstLoad') |  socket.emit('config', config);  |
|  socket.on('stop')  |  socket.emit("firmware", firmware); |
|  socket.on('pause') |  socket.emit("firmware", firmware); |
|  socket.on('unpause') |  socket.emit("activePorts", port.path + ',' + port.options.baudRate); (not even sure we really use this) |
| socket.on('serialSend')  |  socket.emit("connectStatus", 'strings here'); |
| socket.on('feedOverride')  |   |
| socket.on('spindleOverride') |   |
| socket.on('laserTest')  |   |
| socket.on('clearAlarm') |   |
| socket.on('getFirmware') |   |
| socket.on('refreshPorts') |   |
| socket.on('closePort') |   |
| socket.on('areWeLive') |   |
| socket.on('connectTo') |   |

openhardwarecoza, 2016-12-19:
Moved comms discussion to https://github.com/LaserWeb/LaserWeb4/issues/92