LaserWeb/LaserWeb4 #90: [Help] Creating an Electron Branch

jorgerobles, 2016-12-18:
Hi, due the ReGL issues (who I am lucky and not being affected yet) I'm trying to make an Electron Build (Chromium 53)... without luck. 

I was following this guide https://medium.com/@ad_harmonium/build-to-both-electron-js-and-browser-targets-using-webpack-59266bdb76a, but got stuck on when thrown several errors regarding ES6 transliteration.
Then I went for https://github.com/paulcbetts/electron-prebuilt-compile without luck.

Any Electron savvy out there?

openhardwarecoza, 2016-12-18:
@montekrol @copet80

openhardwarecoza, 2016-12-18:
Not webpack,  but maybe helpful https://github.com/LaserWeb/LaserWeb3-Electron

jorgerobles, 2016-12-18:
Thanks @openhardwarecoza already check LW3 branch but seems webpack does things its own way.

kaosat-dev, 2016-12-18:
Can you share the errors ? I would love to help out on this.

jorgerobles, 2016-12-18:
I will post tomorrow @kaosat-dev, gonna try a fresh start :)

kaosat-dev, 2016-12-18:
@jorgerobles  :) ok

openhardwarecoza, 2016-12-19:
(; Skip the seperate branch

Make the Electron app serve two purposes:  

Web server serving /dist/  and also the Electron UI

The reason for this is that LW is always nice to access over the LAN from another PC.  With the current LW3 branch, the app is only accessible via the electron gui's chrome.  node-static/http doesnt play nice with Electron (Open issue here https://github.com/electron-userland/electron-builder/issues/974 ) once built into an exe.  

This way, the Electron app can be used, or while its running, we can open a browser to localhost:8080 or from another machine on the lan

We are definitely going electron (about a month of fights in the LW3 branch determined it works well enough (though real Chrome was faster than Electron, so therefore an even bigger wish to get the webserver going too))

openhardwarecoza, 2016-12-19:
@jorgerobles : done see https://github.com/LaserWeb/LaserWeb4/issues/79#issuecomment-267953205

closed 2016-12-19 by openhardwarecoza

kaosat-dev, 2016-12-19:
@openhardwarecoza actually that is a different thing I believe, the one in #79 is merely a test for regl /os issues,  it does not use webpack or anything specific to LW4 so I doubt It can be use for this issue :)

jorgerobles, 2016-12-19:
Indeed. I'm struggling to get a webpack-react-electron-app-of-hell. I'm sure a quantum dynamics pHd is easier.

openhardwarecoza, 2016-12-19:
Run the distribution build,  copy /dist and host that?

jorgerobles, 2016-12-19:
Something interferes in the middle. Its the webpack middleware or Hot..
I've lot the lead.
I'm now trying from scratch, with a test project to reproduce the issue.
All the tutorials I've found tell this is easy, hence, I'm doing something
wrong. :)

openhardwarecoza, 2016-12-19:
> issuecomment-268022559

I ran /dist/ from another webserver and it worked

jorgerobles, 2016-12-19:
I'll take into account 👍