LaserWeb/LaserWeb4 #92: Work on Comms module

openhardwarecoza, 2016-12-19:


jorgerobles, 2016-12-19:
@cprezzi Although react could be used to develop on serverside (in theory) there's no need to use it, so start a cozy server with a well defined api and we will connect to it without problem.

I think the one that has a solid idea of how to use react is @tbfleming. I'm rather a well fed monkey.

openhardwarecoza, 2016-12-19:
Agreed on keeping server like we have now.  Then we can also reuse the existing electron packing

cprezzi, 2016-12-19:
Ok, I'll go for it. 
I think it makes sense to emit a result for most client commands so the clients can react on it, like change button "Pause" to "Play" (instead of guessing the state on it's own).

openhardwarecoza, 2016-12-19:
Indeed it makes more sense!

openhardwarecoza, 2016-12-20:
Note Keep this issue open till https://github.com/LaserWeb/lw.comm-server is nearer ready

closed 2017-01-13 by openhardwarecoza